Peter-van-Tol/LiteX-CNC #12: pip install does not work on debian 10, works on arch linux?

cdwijs, 2023-01-21:
"pip install --extra-index-url https--test.pypi.org/simple/ litexcnc[cli]" does not work on linuxCNC:
`
cedric@virt-cnc:~$ sudo -i
[sudo] password for cedric:
root@virt-cnc:~# useradd -m testuser
root@virt-cnc:~# su testuser
$ bash --login
testuser@virt-cnc:/root$ cd
testuser@virt-cnc:~$ git clone https://github.com/Peter-van-Tol/LiteX-CNC.git
Cloning into 'LiteX-CNC'...
remote: Enumerating objects: 1259, done.
remote: Counting objects: 100% (421/421), done.
remote: Compressing objects: 100% (168/168), done.
remote: Total 1259 (delta 293), reused 313 (delta 230), pack-reused 838
Receiving objects: 100% (1259/1259), 4.03 MiB | 3.66 MiB/s, done.
Resolving deltas: 100% (795/795), done.
testuser@virt-cnc:~$ cd LiteX-CNC/
testuser@virt-cnc:~/LiteX-CNC$ pip install --extra-index-url https://test.pypi.org/simple/ litexcnc[cli]
Looking in indexes: https://pypi.org/simple, https://test.pypi.org/simple/
Collecting litexcnc[cli]
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/litexcnc/
`
My versions:
`
testuser@virt-cnc:~/LiteX-CNC$ pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)
testuser@virt-cnc:~/LiteX-CNC$ python --version
Python 2.7.16
testuser@virt-cnc:~/LiteX-CNC$ linuxcnc -v
LINUXCNC - 2.8.4-1-gb7824717b
testuser@virt-cnc:~/LiteX-CNC$ uname -a
Linux virt-cnc 4.19.0-21-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.249-2 (2022-06-30) x86_64 GNU/Linux
testuser@virt-cnc:~/LiteX-CNC$ cat /etc/issue
Debian GNU/Linux 10
`

cdwijs, 2023-01-21:
"pip install --extra-index-url https--test.pypi.org/simple/ litexcnc[cli]" does work on linuxCNC:
`
[cedric@cedric ~]$ su
Password: 
[root@cedric cedric]# useradd -m testuser
[root@cedric cedric]# passwd testuser
New password: 
Retype new password: 
passwd: password updated successfully
[root@cedric cedric]# su testuser
[testuser@cedric cedric]$ bash --login
[testuser@cedric cedric]$ cd
[testuser@cedric ~]$ git clone https://github.com/Peter-van-Tol/LiteX-CNC.git
Cloning into 'LiteX-CNC'...
The authenticity of host 'github.com (140.82.121.3)' can't be established.
ED25519 key fingerprint is SHA256:+DiY3wvvV6TuJJhbpZisF/zLDA0zPMSvHdkr4UvCOqU.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
Cloning into 'LiteX-CNC'...
remote: Enumerating objects: 1259, done.
remote: Counting objects: 100% (421/421), done.
remote: Compressing objects: 100% (168/168), done.
remote: Total 1259 (delta 293), reused 313 (delta 230), pack-reused 838
Receiving objects: 100% (1259/1259), 4.03 MiB | 3.70 MiB/s, done.
Resolving deltas: 100% (795/795), done.
[testuser@cedric ~]$ cd LiteX-CNC/
[testuser@cedric LiteX-CNC]$ pip install --extra-index-url https://test.pypi.org/simple/ litexcnc[cli]
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://test.pypi.org/simple/
Collecting litexcnc[cli]
  Downloading https://test-files.pythonhosted.org/packages/7a/e8/59f65e8b7ffb00e6d336cc72845a9162d85064497d42c3b815adbc132cc2/litexcnc-0.9.0a5-py3-none-any.whl (168 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 168.9/168.9 kB 2.7 MB/s eta 0:00:00
Requirement already satisfied: pydantic<2.0.0,>=1.10.2 in /usr/lib/python3.10/site-packages (from litexcnc[cli]) (1.10.4)
Requirement already satisfied: packaging<22.0,>=21.3 in /usr/lib/python3.10/site-packages (from litexcnc[cli]) (21.3)
Requirement already satisfied: click<9.0.0,>=8.1.3 in /usr/lib/python3.10/site-packages (from litexcnc[cli]) (8.1.3)
Collecting ninja<2.0.0,>=1.11.1
  Downloading ninja-1.11.1-py2.py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (145 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 146.0/146.0 kB 4.4 MB/s eta 0:00:00
Requirement already satisfied: requests<3.0.0,>=2.28.1 in /usr/lib/python3.10/site-packages (from litexcnc[cli]) (2.28.1)
Collecting yapps<3.0.0,>=2.2.0
  Downloading Yapps-2.2.0.tar.gz (13 kB)
  Preparing metadata (setup.py) ... done
Collecting meson<0.65.0,>=0.64.1
  Downloading meson-0.64.1-py3-none-any.whl (895 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 895.6/895.6 kB 5.5 MB/s eta 0:00:00
Requirement already satisfied: chardet<6.0.0,>=5.0.0 in /usr/lib/python3.10/site-packages (from litexcnc[cli]) (5.1.0)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /usr/lib/python3.10/site-packages (from packaging<22.0,>=21.3->litexcnc[cli]) (3.0.9)
Requirement already satisfied: typing-extensions>=4.2.0 in /usr/lib/python3.10/site-packages (from pydantic<2.0.0,>=1.10.2->litexcnc[cli]) (4.4.0)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3.10/site-packages (from requests<3.0.0,>=2.28.1->litexcnc[cli]) (3.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3.10/site-packages (from requests<3.0.0,>=2.28.1->litexcnc[cli]) (1.26.12)
Building wheels for collected packages: yapps
  Building wheel for yapps (setup.py) ... done
  Created wheel for yapps: filename=Yapps-2.2.0-py3-none-any.whl size=15816 sha256=d5731fa64e909906d5d47cdf3fa017b739a12de66ec5f3e092948ed35ea0e9bb
  Stored in directory: /home/testuser/.cache/pip/wheels/f4/b6/0c/c902559099e80496ae3917c39c7964ad092ab64f5ba9882e0a
Successfully built yapps
Installing collected packages: yapps, ninja, meson, litexcnc
  WARNING: The script yapps2 is installed in '/home/testuser/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script ninja is installed in '/home/testuser/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script meson is installed in '/home/testuser/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script litexcnc is installed in '/home/testuser/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed litexcnc-0.9.0a5 meson-0.64.1 ninja-1.11.1 yapps-2.2.0
`
My versions:
`
[testuser@cedric LiteX-CNC]$ pip --version
pip 22.3.1 from /usr/lib/python3.10/site-packages/pip (python 3.10)
[testuser@cedric LiteX-CNC]$ python --version
Python 3.10.9
[testuser@cedric LiteX-CNC]$ linuxcnc -v
bash: linuxcnc: command not found
[testuser@cedric LiteX-CNC]$ uname -a
Linux cedric 6.1.4-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 07 Jan 2023 15:10:07 +0000 x86_64 GNU/Linux
`

Peter-van-Tol, 2023-01-21:
On Debian pip reverts to Python 2.7, which is not supported. To install on Debian, please try:
    `pip3 install --extra-index-url` https://test.pypi.org/simple/ litexcnc[cli]

Peter-van-Tol, 2023-07-12:
Solved in latest documentation:
https://litex-cnc.readthedocs.io/en/11-add-external-extensions-to-litexcnc/readme.html#installation

closed 2023-07-12 by Peter-van-Tol