Peter-van-Tol/LiteX-CNC #13: pip3 install gives error 404

dixi83, 2023-04-01:
**Describe the bug**
the command `pip3 install --extra-index-url https://test.pypi.org/simple/ litexcnc[cli]` gives the folowing error:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://test.pypi.org/simple/
Collecting litexcnc[cli]
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/litexcnc/
```

**To Reproduce**
`pip3 install --extra-index-url https://test.pypi.org/simple/ litexcnc[cli]` and enter ;)

**Expected behavior**
Installation of LitexCNC

**LinuxCNC:**
 - OS:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```
 - Version 2.8.4

**Additional context**
-

Peter-van-Tol, 2023-04-03:
Thank you for posting the bug. I've verified the command you've used: they are correct. According to [this bug report](https://github.com/pypa/pip/issues/7486) i9t is an issue with the operating system.

Could you try to update `pip` by using `pip3 install -U pip` to ensure you have the latest version of `pip`.

Peter-van-Tol, 2023-04-03:
Alternatively, you can also built this package from source using `poetry`. If you are willing to do this, I can recommend you doing this from #11:
`pip install -e "litexcnc[cli] @ git+https://github.com/Peter-van-Tol/LiteX-CNC.git@11-add-external-extensions-to-litexcnc"`

dixi83, 2023-04-03:
Thanks peter I'll test it probably next Friday.

dixi83, 2023-04-08:
After a little struggle I managed to install it ;)
The command you give me was giving a 
`ERROR: litexcnc[cli] @ git+https:  .......... is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with bzr+http, bzr+https, bzr+ssh, bzr+sftp, bzr+ftp, bzr+lp, bzr+file, git+http, git+https, git+ssh, git+git, git+file, hg+file, hg+http, hg+https, hg+ssh, hg+static-http, svn+ssh, svn+http, svn+https, svn+svn, svn+file).`

I did a bit Python coding but I never got into building packages or go in depth with the pip package manager... 

My Install log:
```
$: git clone https://github.com/Peter-van-Tol/LiteX-CNC.git
$: cd LiteX-CNC
$: git checkout 11-add-external-extensions-to-litexcnc
$: pip install ./ litexcnc[cli]
```

Maybe is helps someone else... This issue can be closed. I'll continue my journey ;)

P.S. today I got my 5A-75E (v8.2) and the 74LVC245 IC's! Lets hope the 8.2 version will not get me in trouble..

closed 2023-04-08 by dixi83