Peter-van-Tol/LiteX-CNC #14: RV901T how to ?

tuxcnc, 2023-04-27:
There is wrote "LitexCNC can be used to create the Verilog" but no explanation how do it.

Peter-van-Tol, 2023-04-27:
At this moment I'm enjoying my holiday. I will post here coming week a workflow for creating the Verilog. If you want to get started, you can already start downloading and installing Xilinx for the target architecture of the RV901T. Given the size, it took me a while to get the development environment up and running

dlhenke, 2023-04-27:
see this thread at linuxcnc forum 
https://forum.linuxcnc.org/27-driver-boards/47730-new-project-litehm2-a-hostmot2-port-to-linsn-rv901t

tuxcnc, 2023-04-28:
> At this moment I'm enjoying my holiday. I will post here coming week a workflow for creating the Verilog. If you want to get started, you can already start downloading and installing Xilinx for the [...]

I will be patient, I purchased RV901T on Aliexpress and  waiting for delivery.
I have Xilinx_ISE_DS_14.7 installed yet.
Thanks in advance.

tuxcnc, 2023-05-07:
I got it, I suppose...
The problem was in examples/rv901t_emco5.json file.
There are many errors in "stepgen" section, most important is "dir_pin": "J600:231" that causes error with useless message "KeyError: 231"...
After repair the json file i got Verilog file, but not tested in Xilinx ISE yet.

Peter-van-Tol, 2023-05-07:
As soon you get a valid Verilog file with the command like:
`litexcnc build_firmware <config.json> --alias-file` 
you should be able to get a `.v`-file and import the at in Xilinx ISE. 

When you import the file,you'll get some errors about constraints not being declared (?). This is a bug in Litex, and not due to an error in the firmware. The solution is to remove all the lines where an error occurs, then compile it.

tuxcnc, 2023-05-27:
No succes.
The ISE 14.7 works properly.
I can compile verilog sources, load to FPGA or load to flash, by Xilinx Platform Cable.
The programs like blink example works.
"litex build-firmware" produced verilog file before fails.
ISE produces tons of warnings (no error) like "signal not connected", but bitstream is compiled.
After load to FPGA, board not responds to ping.
I know, it is possible to compile working bitstream, but no idea how.

closed 2023-06-07 by Peter-van-Tol