Peter-van-Tol/LiteX-CNC #15: install_driver error

OJthe123, 2023-05-11:
I installed litexcnc additional extensions branch with poetry.
when I do 'poetry run litexcnc install_driver' it throws errors when compiling litexcnc.c

> $ poetry run litexcnc install_driver
> INFO: Retrieving default driver files to compile...
> Copying file 'watchdog.c'
> Copying file 'wallclock.c'
> Copying file 'litexcnc.c'
> Copying file 'litexcnc_encoder.c'
> Copying file 'litexcnc_stepgen.c'
> Copying file 'litexcnc_gpio.c'
> Copying file 'litexcnc_pwm.c'
> Copying file 'pos2vel.c'
> Copying file 'etherbone.c'
> Copying file 'litexcnc_eth.c'
> Copying file 'litexcnc.h'
> Copying file 'wallclock.h'
> Copying file 'watchdog.h'
> Copying file 'litexcnc_encoder.h'
> Copying file 'litexcnc_stepgen.h'
> Copying file 'litexcnc_pwm.h'
> Copying file 'litexcnc_gpio.h'
> Copying file 'pos2vel.h'
> Copying file 'etherbone.h'
> Copying file 'litexcnc_eth.h'
> INFO: Compiling LitexCNC driver...
> Compiling realtime litexcnc.c
> litexcnc.c: In function ‘litexcnc_register’:
> litexcnc.c:512:1: warning: label ‘fail0’ defined but not used [-Wunused-label]
>   512 | fail0:
>       | ^~~~~
> litexcnc.c: In function ‘register_module’:
> litexcnc.c:539:22: error: ‘LINELEN’ undeclared (first use in this function)
>   539 |     char module_path[LINELEN+1];
>       |                      ^~~~~~~
> litexcnc.c:539:22: note: each undeclared identifier is reported only once for each function it appears in
> litexcnc.c:548:10: warning: unused variable ‘register_name’ [-Wunused-variable]
>   548 |     char register_name[LINELEN+1];
>       |          ^~~~~~~~~~~~~
> litexcnc.c:539:10: warning: unused variable ‘module_path’ [-Wunused-variable]
>   539 |     char module_path[LINELEN+1];
>       |          ^~~~~~~~~~~
> litexcnc.c: In function ‘register_driver’:
> litexcnc.c:574:22: error: ‘LINELEN’ undeclared (first use in this function)
>   574 |     char driver_path[LINELEN+1];
>       |                      ^~~~~~~
> litexcnc.c:583:10: warning: unused variable ‘register_name’ [-Wunused-variable]
>   583 |     char register_name[LINELEN+1];
>       |          ^~~~~~~~~~~~~
> litexcnc.c:574:10: warning: unused variable ‘driver_path’ [-Wunused-variable]
>   574 |     char driver_path[LINELEN+1];
>       |          ^~~~~~~~~~~
> make: *** [/usr/share/linuxcnc/Makefile.modinc:115: litexcnc.o] Fehler 1
> Error: Compilation of the driver failed.

is it the poetry environment that causes this problem? linking problems? 
or is there a missing declaration of LINELEN?

tuxcnc, 2023-05-18:
> I did a new install in a virtual maschine with Debian 12 and Linuxcnc 2.9
> 
> EDIT: I installed it with
> 
> > $: git clone https://github.com/Peter-van-Tol/LiteX-CNC.git
> > $: cd LiteX-CNC
> > $: git checkout 11-add-external-extensions-to-litexcnc
> > $: pip3 install ./ litexcnc[cli]
> 
> but the error is still there

I got this same.
Do NOT git checkout 11-add-external-extensions-to-litexcnc.

Peter-van-Tol, 2023-05-22:
For my machine I installed it with:
``` shell
pip install litexcnc[cli] @ git+https://github.com/Peter-van-Tol/LiteX-CNC@11-add-external-extensions-to-litexcnc
```

My machine works with LinuxCNC 2.8.2, so most likely LINELEN is not defined in the latest version. Will patch this in #11 as soon as I know where this define went.

Peter-van-Tol, 2023-05-22:
The definition of LINELEN  was moved between 2.8 and 2.9. I've patch the driver for this omission in #11 .

closed 2023-05-22 by Peter-van-Tol