Peter-van-Tol/LiteX-CNC #17: Differential outputs

tuxcnc, 2023-05-19:
I think this is not a big problem to add to Stepgen and PWM negated output signals.
For example:
"step_pin": "J9: 0",
"step_pin_inv": "j9: 1",
This would give the possibility to control the output devices with a differential signal.
This won't spoil anything, if someone doesn't need it, they just won't use it.

OJthe123, 2023-05-21:
I think this is already in progress, because the breakoutboards have the differential option
https://github.com/Peter-van-Tol/HUB-75-boards

tuxcnc, 2023-05-21:
> I think this is already in progress, because the breakoutboards have the differential option https://github.com/Peter-van-Tol/HUB-75-boards

Evidently, the author started from the end ...

kubabuda, 2023-05-21:
Why do you want two pins per diff line? Transceiver chip on daughter board normally handles this for you. You want to drive differential bus like RS422 straight from FPGA IO (without all protection in transceiver)?

kubabuda, 2023-05-22:
@OJthe123 thats exactly my point

Peter-van-Tol, 2023-05-22:
The differential stepgen is available from `Litex-CNC`. The way it is implemented in the JSON-configuration is:
```json
"modules": [
    "modules": [
        ...,
        {
            "module_type": "stepgen",
            "instances": [
                {
                    "pins" : {
                        "stepgen_type": "step_dir_differential",
                        "step_pos_pin": "j9:6",
                        "step_neg_pin": "j9:5",
                        "dir_pos_pin": "j9:4",
                        "dir_neg_pin": "j9:2"
                    },
                    "soft_stop": true
                },
                ...
            ]
        }
    ]
```

To stand corrected: the module has been developed first and then the BOB. 

The BOB outputs 0 and 5 Volt, analog to the encoder I had laying around. If this is considered to be not 'true' differential, one can also opt to use the stepgen_type `step_dir` and use the suggester `tranceiver`. But for 400 kHz pulse train this seems overkill in my opinion.

My rationale for using 2 pins for each signal of the stepper is the abundance of outputs available. My machine (EMCO5 CNC) is almost finished, and still ample outputs left over. If anybody sees an opportunity to contribute in the development of BOB's, please do so for the community. Improvements are welcome!

OJthe123, 2023-05-22:
is the module in the "standard" LiteX-CNC branch? Or in "add-external-extensions" ?

closed 2023-06-07 by Peter-van-Tol