Peter-van-Tol/LiteX-CNC #18: Bad encoder code

tuxcnc, 2023-06-03:
I can't compile configuration with encoder module.
The error message is :

`Traceback (most recent call last):
  File "/usr/local/bin/litexcnc", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/litexcnc/cli/build_firmware.py", line 58, in cli
    builder.build(run=build)
  File "/opt/litex/litex/litex/soc/integration/builder.py", line 332, in build
    self.soc.finalize()
  File "/opt/litex/litex/litex/soc/integration/soc.py", line 1215, in finalize
    self.csr_bankarray = csr_bus.CSRBankArray(self,
  File "/opt/litex/litex/litex/soc/interconnect/csr_bus.py", line 213, in __init__
    self.scan(ifargs, ifkwargs)
  File "/opt/litex/litex/litex/soc/interconnect/csr_bus.py", line 271, in scan
    rmap = CSRBank(csrs, mapaddr,
  File "/opt/litex/litex/litex/soc/interconnect/csr_bus.py", line 176, in __init__
    csr.GenericBank.__init__(self,
  File "/opt/litex/litex/litex/soc/interconnect/csr.py", line 578, in __init__
    c.finalize(busword, ordering)
  File "/opt/litex/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/opt/litex/litex/litex/soc/interconnect/csr.py", line 311, in do_finalize
    nwords = (self.size + busword - 1)//busword
TypeError: can only concatenate tuple (not "int") to tuple
`

The most important line in /opt/litex/litex/litex/soc/interconnect/csr.py is :

`nwords = (self.size + busword - 1)//busword`

The // (floor division) of course operates on numbers.
It seems the fuction is invoked with tuple type variable as "bussword".
It may be a typo in source, but I can't find where from do_finalize function is invoked, and where "bussword" variable is declared.

Peter-van-Tol, 2023-06-03:
This error stems from outside of LitexCNC. I do think it is caused by the way the encoder is added. Could you share your configuration, so I can try to reproduce and compare with working code?

If it is in the config, I will add validation to prevent this in future.

tuxcnc, 2023-06-03:
> This error stems from outside of LitexCNC. I do think it is caused by the way the encoder is added. Could you share your configuration, so I can try to reproduce and compare with working code?

This is your configuration from https://github.com/Peter-van-Tol/LiteX-CNC/issues/16#issuecomment-1560773172

I described problem in that topics, gave you what you ask for, but now I see that you've never read that...

You can shorten the config to : 
`{
    "board_name": "EMCO5",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 35000000,

    "modules": [
        {
            "module_type": "encoder",
            "instances": [
                {"pin_A":"j1:4", "pin_B":"j1:2"}
            ]
        }
    ]
}`

tuxcnc, 2023-06-03:
I'm sorry, I cant "add code" properly, and cant edit it later.
Maybe it is my browser problem.

Without "code" it is  : 

{
    "board_name": "EMCO5",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 35000000,

    "modules": [
        {
            "module_type": "encoder",
            "instances": [
                {"pin_A":"j1:4", "pin_B":"j1:3"}
            ]
        }
    ]
}