Peter-van-Tol/LiteX-CNC #19: New cofig and test files.

tuxcnc, 2023-06-04:
The main repository contains obsolete and abandoned version, so to do any with that is waste time.
The  11-add-external-extensions-to-litexcnc contains new version, but without any documentation, so even if something works, only @Peter-van-Tol  know how to run this... 
The minimum is one example config file with one module of each type, and one example hal file, even with one axis. 
But tested, not written from memory, and with typos and bugs.
Without this nobody can test new version.

Peter-van-Tol, 2023-06-04:
The documentation is (and has been) online:
https://litex-cnc.readthedocs.io/en/11-add-external-extensions-to-litexcnc/

tuxcnc, 2023-06-04:
> The documentation is (and has been) online: https://litex-cnc.readthedocs.io/en/11-add-external-extensions-to-litexcnc/

Are you sure ?

I tried today stepgen test. 
I can't run all code, but LinuxCNC loads enought for I run halmeter, so I can see what modules are available.
The stepgen example from your link  is wrong, contains not working functions and not existing pins...

[image]
[image]

Peter-van-Tol, 2023-06-04:
The example is based on the `stepgen` example from LinuxCNC. Please note that a lot of the pins which do not exist in your list are actually parameters, so they will be listed 'Parametry' in your example.

If you want to be helped:
- please be kind, I have also a family, a job and some more projects on my hand;
- please be concise. If something is wrong, please print out the error messages (if any) or say which pins / params do not exist. It saves me a lot of work, work I could have put in developing this project further.

closed 2023-06-07 by Peter-van-Tol