Peter-van-Tol/LiteX-CNC #21: Velocity is inverted

romanetz, 2023-07-01:
With the software from "main" branch, the velocity is inverted and direction is decreased instead of increas

Set stepgen scale to 1000 pulses per millimeter and max working speed to 15 m/min. When the speed is over 6 m/min, say, 10 m/min, direction is inverted and actual speed differs from commanded speed by sign

Peter-van-Tol, 2023-07-04:
Thanks for this issue! I'm going to look into it, but suspect it is due to an overflow situation. Using this comment as first idea:
- 1000 pulses/mm and 15 m/min results in 15,000,000 pulses per minute or 250 kHz pulse train. This should be in the limit of the design of the FPGA (400 kHz).
- The actual error is reported to occur much earlier at 6 m/min. This results in the error occurring at 100 kHz.

The following code is the culprit:
``` c
shift = 0
while (soc.clock_frequency / (1 << shift) > 400e3):
    shift += 1
```

The idea was that the maximum step rate would be 400 kHz. But, the shift is off by one, so this will allow for a guaranteed step rate of 200 kHz. If in your case the FPGA clock speed is 30 MHz, your symptoms are completely explained by this mistake. This mistake affects both `main` and `11-add-external-extensions-to-litexcnc` branches.

Proposed solution is to change above code with:
``` c
shift = 0
while (soc.clock_frequency / (1 << (shift + 1)) > 400e3):
    shift += 1
```

Peter-van-Tol, 2023-07-04:
Also just noticed that there is no safeguard against this. When maximum speed is defined in the stepgen parameters, these will be obeyed. Even when they are above the theoretical maximum of the FPGA. The maximum speed should be limited to the maximum theoretical speed of the FPGA.

The code for getting the max frequency is:
```c
 stepgen->data.max_frequency = (double) (*(stepgen->data.clock_frequency)) / (steplen_cycles + stepspace_cycles);
```

This should be the minium of:
- the above formula, which takes the timing into account
- the max frequency as set by the integer value which is sent to the FPGA

Peter-van-Tol, 2023-07-04:
I think these changes can be implemented this week,

romanetz, 2023-07-07:
I changed firmware in FPGA and driver like this:
shift = 0
while (soc.clock_frequency / (1 << (shift + 1)) > 1000e3):
    shift += 1
This worked for me. Thanks!

Peter-van-Tol, 2023-07-07:
In your code I see you both changed `shift` to `shift + 1` and the maximum frequency to 400. Is the value of 1000 strictly necessary for the correct working?

romanetz, 2023-07-08:
My intention was to achieve 1 m/sec speed with resolution of 1000 steps/mm

Peter-van-Tol, 2023-07-12:
I've resolved this bu in both branches (`main` and `11`), could you please try whether the latest version will allow you running at the desired speed?

The value of 1000e3 in your modified code would lead to a max step frequency of 1 MHz. The required speed is 16 kHz (1000 * 1000 / 16), so the default value of 400 kHz is safe. To gain a little on precision/resolution, you can consider to modify the new parameter `<card-name>.stepgen.max-driver-freq` to a lower value, for example 100000 (i.e. 100 kHz).

romanetz, 2023-07-17:
The issue is fixed

closed 2023-07-17 by romanetz

Peter-van-Tol, 2023-07-17:
Thank you for your feedback.