Peter-van-Tol/LiteX-CNC #22: 11-add-external-extensions-to-linuxcnc litexcnc.c

ozzyrob, 2023-07-07:
Should the following code
```
// Between version 2.8 and 2.9 the definition of LINELEN has moved. Below
// is a check whether 2.9 is installed. If so, the header is imported to
// get the correct value of LINELEN.
#ifdef LINUXCNC_H
#include "linuxcnc.h"
#endif /* LINUXCNC_H */

#include "litexcnc.h"
```

be 
```
// Between version 2.8 and 2.9 the definition of LINELEN has moved. Below
// is a check whether 2.9 is installed. If so, the header is imported to
// get the correct value of LINELEN.
#ifndef LINUXCNC_H
#include "linuxcnc.h"
#endif /* LINUXCNC_H */

#include "litexcnc.h"
```

When compiling with 2.9 the first doesn't work but the second will.

Also while I have it in my head:
`sudo .platformio/penv/bin/litexcnc install_driver`
I've found is required to install the driver. Is this the behavior you wanted. Found both on Mint 21 & Debian Bookworm

There is another issue I found when building for 5a-75e-V7.1 but I will bring this up seperately

Peter-van-Tol, 2023-07-07:
Your correct that the code does nor work with LinuxCNC 2.9. However, when using this code on LinuxCNC 2.8 it won't work, as the import "linuxcnc.h" does not exist. Have to investigate how to check for this with the pre-processor.

Peter-van-Tol, 2023-07-07:
Could you please verify that the following code works on LinuxCNC 2.9?
```c
// Between version 2.8 and 2.9 the definition of LINELEN has moved. Below
// is a check whether 2.9 is installed. If so, the header is imported to
// get the correct value of LINELEN.
#ifndef LINELEN
#include "linuxcnc.h"
#endif /* LINELEN*/
```

ozzyrob, 2023-07-08:
Can confirm, as that is the code mod I used in the opening post.

Peter-van-Tol, 2023-07-09:
Your change has been applied to 11-add-external-extensions-to-linuxcnc. Thanks a lot!

closed 2023-07-09 by Peter-van-Tol