Peter-van-Tol/LiteX-CNC #26: Rewite to Add files via upload (PR #25)

ozzyrob, 2023-07-16:
Hi Pete,
I was looking at the sed thing, it's not the most readable bit of code, and had a go at making it more "python-like". Now I'm not very experienced at python, actually this is my first real go at it. Anyways I came up with the following code.  I was just wondering would this be a better & more readable solution. It also saves the preprocessor from making the decision from deciding which path to include. It still gets the correct path for Makefile.modinc and then decides which value for the rtlibdir variable.
I'm hoping that it would it make it easier in the future rather than that monolithic sed command.

```

import shutil
import subprocess
import re

def index_containing_substring(the_list, substring, start=0):
    #for i, s in enumerate(the_list,start):
    for i in range(start, len(the_list)):
        if substring in the_list[i]:
            #print(i,the_list[i])
            return i
    return -1

def get_rtlibdir():
    halcompile = shutil.which('halcompile')
    if not halcompile:
            raise SystemExit("Unable to locate halcompile, is it installed?")
    try:
        modinc_loc = subprocess.run(["halcompile --print-modinc"], check = True, shell=True, capture_output=True, text=True)
    except subprocess.CalledProcessError:
        raise SystemExit("halcompile --print-modinc failed")

    modinc_path = (modinc_loc.stdout).rstrip()
    f = open(f'{modinc_path}','r')
    lines = f.read().splitlines() # List with stripped line-breaks
    f.close()                     # Close file

    index = index_containing_substring (lines, 'RUN_IN_PLACE=')
    rip_tmp = re.sub(' ', '',lines[index]).partition('=')

    index = index_containing_substring (lines, 'RTLIBDIR :=')
    rip_tmp = re.sub(' ', '',lines[index]).partition('=')
    rip_yes = rip_tmp[2]

    index = index_containing_substring (lines, 'RTLIBDIR :=',index+2)
    rip_tmp=re.sub(' ', '',lines[index]).partition('=')
    rip_no = rip_tmp[2]

    if rip_tmp[2] == 'yes':
        return rip_yes
    else:
        return rip_no

rtlib = get_rtlibdir()
print (f"{rtlib}")
# from this section forward we can use the original code from install_driver.py
# this was just added to check the code written to out config.h
f = open('config.h','w')
print("/**", file=f)
print(" * THIS FILE IS AUTOGENERATED BY HALCOMPILE.PY, CHANGES WILL BE OVERWRITTEN", file=f)
print(" */", file=f)
print("#ifndef __INCLUDE_LITEXCNC_CONFIG_H__", file=f)
print("#define __INCLUDE_LITEXCNC_CONFIG_H__", file=f)
print("", file=f)
print(f"#define EMC2_RTLIB_DIR \"{rtlib}\"", file=f)
print("", file=f)
print("#endif /* __INCLUDE_LITEXCNC_CONFIG_H__ */", file=f)
f.close()

```

Peter-van-Tol, 2023-07-17:
To be honest, the `sed` thing was indeed black magic. I accepted your PR because it was working, and tested it with just running the command.

Now we have something working, something maintainable is even better. At this moment I'm enjoying a holiday (without laptop or CNC machine), as soon as I'll be back I'm going to check this code, test it and merge it. I really appreciate your effort to get LitexCNC working and better.

Peter-van-Tol, 2023-08-02:
I have merged the pull request, thanks for your effort.

closed 2023-08-02 by Peter-van-Tol