Peter-van-Tol/LiteX-CNC #28: starting LinuxCNC fails with new driver

OJthe123, 2023-07-20:
Hi.
I did "git pull" to get the updated files from 11-add-external....

then installed the new drivers "litexcnc install_driver"
rebuild the firmware.

Then LinuxCNC won't start anymore.
[linuxcnc-report.txt](https://github.com/Peter-van-Tol/LiteX-CNC/files/12103121/linuxcnc-report.txt)

ozzyrob, 2023-07-20:
This is the interesting bit on line #80
Waiting for component 'inihal' to become ready......................................A configuration error is preventing LinuxCNC from starting.

It's really hard to debug without all the ini & hal files.

Peter-van-Tol, 2023-07-20:
Thank you for your response @ozzyrob. It is indeed a configuration error, because the card does connect and buffers are being allocated and after all cleaned up.

Please share your ini and Hal files.

OJthe123, 2023-07-20:
Sure. I should have attached the files...

The thing is, I didn't change anything.
You say you only did bug fix for overflow in the stepgen and something for RIP install?
So, that should have nothing to do with my hal and ini, or?
[Semse.hal.txt](https://github.com/Peter-van-Tol/LiteX-CNC/files/12107620/Semse.hal.txt)
[Semse.ini.txt](https://github.com/Peter-van-Tol/LiteX-CNC/files/12107621/Semse.ini.txt)

Peter-van-Tol, 2023-07-20:
Let me check one thing. Did you come from the main branch or did you update from branch 11?

If you updated from branch 11: yes only the Rip and overflow have been changed.

If you come from the main branch: the config file should be updated and the firmware rebuilt.

Can you also share you're config json?

At this moment I'm on holiday, as soon as I'm back I will build your config and investigate.

Also if there are any logs available from LinuxCNC, please share. Somewhere I might have the feeling a pin has been renamed somewhere in the process. (yes, that's on me)

OJthe123, 2023-07-20:
Yes. I come from the 11-add-externals.
I used the new json format. It works fine.
Only when I pull the last update I cannot start LinuxCNC

ozzyrob, 2023-07-20:
If you can describe your system, version of Linuxcnc, Debian version & kernel version, and your method of installing Linuxcnc I can try and replicate.
I would also require your json file and the following mentioned in your ini file

HALFILE = custom.hal
POSTGUI_HALFILE = postgui_call_list.hal
SHUTDOWN = shutdown.hal 

If that is ok with you Pete. I'm unemployed ATM and looking to my mind occupied.

OJthe123, 2023-07-21:
All the information should be in the report I attached in the first post above.
I cloned the repo with git clone --single-branch 11-external....
poetry Install
poetry shell
> litexcnc install_driver

Peter-van-Tol, 2023-07-21:
@ozzyrob: I'm completely okay with that when you can try and help others. On the other hand, hope your in a job soon again.

ozzyrob, 2023-07-21:
Cheers Pete

Yeah sorry mate, you're right, was first thing in the morning down under. Just still need the json file so I can build the firmware and flash the fpga.

OJthe123, 2023-07-21:
Sorry.
I am not at home till end of month.
I used the example json.
Just added the index pins to encoders. 3 pwm. 4 stepgens.

ozzyrob, 2023-07-22:
I'm going take a stab at this and suggest it could be a latency issue.  Have you tried  isolcpus=1 when booting ?

The reason I think this I tried a simple sample config that uses the hal_speaker component and as my kernel was non real time I was getting these errors.

waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
USRMOT: ERROR: command timeout

OJthe123, 2023-07-22:
It could be. But as I said, I did not change any other thing.
I can start and use LinuxCNC with LiteX with no problems when I switch back to the "old" drivers

ozzyrob, 2023-07-24:
Can confirm main branch is ok, although I get **Apply time exceeded limits** with 2.9 least linuxcnc loads, haven't tested 2.8 yet.

With 11-add-external on Buster with Linuxcnc 2.8 & Bookworm with Linuxcnc 2.9 sometimes I'm seeing errors as mentioned by OP, sometimes the system is just "freezing".

OJthe123, 2023-07-24:
When I am back at home I will try to go step by step through the components which causes my error

ozzyrob, 2023-07-24:
No probs mate, thanks for all your hard work.

Any thoughts on "Apply Time exceeds limits" on the main branch ?

Peter-van-Tol, 2023-07-24:
The apply time exceeding limits can be due to:
- the first loop taking too much time because every single pre compute is done at the same time, causing the first loop to take too long. Also in my experience the ping time of the card is quiet long. If it happens right at start up, you can ignore this. This behavior can be improved by doing more during the setup phase. This behavior is independent of the version of LinuxCNC. 
- sometimes the network can be lagging. The network calls (especially reading from the card) are blocking.

I did not experience any lock up though on my machines. I'm running on a RPi 4 using isolcpus=1,2,3 for best latency results. Although isolcpus=2,3 also yields good results. Generally it is recommended (no source, this is from the top of my head) to isolate a pair. 

My response time might be higher due to holiday. My apologies for any inconvenience.

ozzyrob, 2023-07-24:
No rush mate, it's all sweet & cruisy Down Under, no need to apologise. Enjoy your holiday.
To be fair I'm testing on Lenovo T530 with a Dual core i5 (it's my favourite, but shhh don't tell my other computers ;) ). The real machine will be a PC with a quad core i5. I just use the laptop as I sit in the living room rather than isolated somewhere else.

So I'll do a bit more testing Tomorrow.

OJthe123, 2023-07-26:
I also have plenty of time with this.
My lathe don't need this Speed when overflow can occur. And I don't use a RIP.
I am fine with the Version that works for me 😄👍

Peter-van-Tol, 2023-08-03:
Back from my holiday and sadly: I can reproduce this bug. Starting my machine leads to freezing of LinuxCNC. Seems that some processes are not running any more.

OJthe123, 2023-08-03:
In the LCnc Forum someone mentioned that this happens when he add stepgens to the Json config

Peter-van-Tol, 2023-08-04:
At this moment I'm thinking my installation is completely broken:
- removed all files from LitexCNC
- created a new machine config from the examples that come with LinuxCNC
- et voila: the real time components fail

Reinstalled LinuxCNC to no avail. Today I'm going to format the image to see whether I can create a working config again....

ozzyrob, 2023-08-04:
You think it's corrupting something in the actual Linuxcnc installation ?

OJthe123, 2023-08-04:
https://www.dropbox.com/s/h1v0j1btdzi96ia/VID_20230804_093236.mp4?dl=0

Hey.
I think it is time to show the world that this project is not only a bugs and feature request 😄

ozzyrob, 2023-08-04:
Looking good.
Have you got any more detail of the X axis ?
I'm trying to come up with something simple for my Myford ML/S7 Frankenstein.

OJthe123, 2023-08-04:
https://www.dropbox.com/s/e3lnlcgugkhil5h/IMG_20230804_094837.jpg?dl=0

https://www.dropbox.com/s/4x0orttlawtoq6c/IMG_20230804_094831.jpg?dl=0

https://www.dropbox.com/s/m911beann6je7yi/IMG_20230804_094825.jpg?dl=0

It is a jmc ihsv57 180w servo with 1204 kus spindle and a selfmade mounting plate

ozzyrob, 2023-08-04:
Cheers, nice solution.

ozzyrob, 2023-08-05:
OK rolled back 11-add-external-extensions-to-litexcnc to: 
```
commit ba57141686940a113f1d2394c17f069025eb3770
Author: Peter van Tol <petervantol@gmail.com>
Date:   Wed Jul 12 10:33:57 2023 +0200

    pip vs. pip3
```
Was able to get the config running, but on a quad core Intel Core i5-3470 with 3 cores isolated I was stil getting
l```
itexcnc: Apply time exceeded limits.litexcnc: Apply time exceeded limits.
Apply time exceeding limits (too long): 69026366277, 69026365867, 69026405879
```

That was with only Linuxnc being run from a terminal.

Peter-van-Tol, 2023-08-05:
@OJthe123 : how would you like the idea of crating a show your machine page in the documentation?

@ozzyrob : this is one of the problems which has been resolved between your rollback version and the current version. Because I'm experiencing the same problem (reinstalling atm), I hope the problem is fixed. Otherwise I'll make a patch for you.

OJthe123, 2023-08-05:
@Peter-van-Tol : Sure, no problem. What do you need?

I also have the "Apply time.." info. But I cannot say that has any impact on  my machine...

OJthe123, 2023-08-05:
Here are my machine files...
[semse.7z.zip](https://github.com/Peter-van-Tol/LiteX-CNC/files/12267116/semse.7z.zip)

Peter-van-Tol, 2023-08-06:
Spend today reinstalling LinuxCNC on my RaspberryPi, bu to no avail. Something has changed apparently and prevents the real-time components to start (i.e. emcTrajInit failed)...

ozzyrob, 2023-08-07:
Any luck yet ?
Sounds like a real PITA. :(

OJthe123, 2023-08-08:
Just for more success story 😄

G76 Threading cycle also works

https://www.dropbox.com/scl/fi/gd01e3tdoovll69d235b6/VID_20230808_141143.mp4?rlkey=5lxysurznfe3msimlakpjrkz7&dl=0

Peter-van-Tol, 2023-08-08:
@OJthe123 : Nice

In the meanwhile, my RPi is showing signs of life again, no errors when starting a simple configuration. Now going to in

ozzyrob, 2023-08-10:
Was just going to mention that the code in litexcnc.c is the same in the commit ba57141686940a113f1d2394c17f069025eb3770. And that works apart from the apply time messages.

Peter-van-Tol, 2023-08-10:
Took some effort, but have found the error. If you pull the latest version of the branch  #11 your LinuxCNC should start up again.

ozzyrob, 2023-08-10:
Fantastic will try in the morning

ozzyrob, 2023-08-11:
Ok gave it a go, tried with the OP's configs.......damned if I could get rid of the following error. Latency is good, I can run a config using steppers with a 25us base thread on this machine. Ping times are good. Tried isolating various cores (4 core i5)

But after that whinge it does start up, just can't jog.

Peter-van-Tol, 2023-08-25:
The following error should be gone by the latest commit. There was a difference between Python (firmware) and C (driver) in determining the pick-off. For slow movements this could be compensated by the PID or `pos2vel`. However, for faster speeds the difference became to big.

The current commit has been tested on my EMCO5, which shows no following error when trying 1500 mm/min whilst using `pos2vel` as translation between position and velocity.

Peter-van-Tol, 2023-08-27:
Continuing from #29 ...

With the config and hal-files from @OJthe123 I can now re-produce the problem. The difference between my setup and his is mainly the scale. Now that has been sorted out, I can start debugging. Just want to close this issue in a proper manner...

I have suggested that the problem might be with using the pin `position-feedback` instead of `position-prediction`, however this did not solve the problem.

Peter-van-Tol, 2023-08-28:
Just committed the changes😃: 
- `pos2vel` has been removed;
- `stepgen` has now position mode (you can select the pin in HAL);
- the documentation has been updated accordingly.

For an advice on position vs pid: if your setup works, there is no need to change. However, the readability and maintainability of the HAL-file will improve when using the position control. The code below is the minimal example for a single axis, which is roughly 50% reduced in size when compared to a solution with `pid` or `pos2vel`.

```
    STEPGEN - X-AXIS
    ########################################################################
    # - Setup of timings
    setp [LITEXCNC](NAME).stepgen.00.position-scale   [JOINT_0]SCALE
    setp [LITEXCNC](NAME).stepgen.00.steplen          5000
    setp [LITEXCNC](NAME).stepgen.00.stepspace        5000
    setp [LITEXCNC](NAME).stepgen.00.dir-hold-time    10000
    setp [LITEXCNC](NAME).stepgen.00.dir-setup-time   10000
    setp [LITEXCNC](NAME).stepgen.00.max-velocity     [JOINT_0]MAX_VELOCITY
    setp [LITEXCNC](NAME).stepgen.00.max-acceleration [JOINT_0]STEPGEN_MAXACCEL
    # setp [LITEXCNC](NAME).stepgen.00.debug 1
    # - Connect velocity command
    net xpos_cmd joint.0.motor-pos-cmd => [LITEXCNC](NAME).stepgen.00.position-cmd
    net xpos_cmd joint.0.motor-pos-fb  <= [LITEXCNC](NAME).stepgen.00.position-prediction
    # - enable the drive
    net xenable joint.0.amp-enable-out => [LITEXCNC](NAME).stepgen.00.enable
```

I would really appreciate if you would test this latest version, so this issue can be closed as resolved.

OJthe123, 2023-08-29:
You rock dude!
changed the drivers and build new firmware.
Tested my setup with 3000mm/min. No errors. Maybe it could be faster, but I did not want to kill my maschine in case of a bug in firmware or driver 😆

ozzyrob, 2023-08-29:
Sorry guys for being a bit quiet, I’ve been playing with some 7c81 firmware on a Spartan 6 dev board. When I get the chance I’ll setup my machine that I use for testing.

ozzyrob, 2023-08-29:
Really Happy Pete. I owe your at least a Beer
Had a play around no following errors, running the sample code was passed, jogging via keyboard passed, MDI passed, only issue was on shutdown. Starting up again runs fine but still quits with the same message.
`
Shutting down and cleaning up LinuxCNC...
task: 48698 cycles, min=0.000007, max=0.097561, avg=0.009917, 0 latency excursions (> 10x expected cycle time of 0.010000s)
litexcnc/Semse: Watchdog timeout not set. Using default value 0 ns (3 times period).litexcnc: LitexCNC etherbone driver unloaded 
rtapi_app: caught signal 11 - dumping core
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Waited 3 seconds for master.  giving up.
Note: Using POSIX realtime
motmod: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Note: Using POSIX realtime
trivkins: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Note: Using POSIX realtime
homemod: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Note: Using POSIX realtime
tpmod: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
<commandline>:0: unloadrt failed
Note: Using POSIX realtime
`

OJthe123, 2023-08-30:
can confirm...
When I run from terminal I can see the same output....still can see no effects on the maschine...

```
Shutting down and cleaning up LinuxCNC...
Running HAL shutdown script
task: 603 cycles, min=0.000041, max=0.012258, avg=0.009716, 0 latency excursions (> 10x expected cycle time of 0.010000s)
mb2hal quit_signal DEBUG: signal [15] received
mb2hal quit_cleanup DEBUG: started
mb2hal quit_cleanup DEBUG: unloading HAL module [16] ret[0]
mb2hal quit_cleanup DEBUG: done OK
mb2hal main OK: going to exit!
litexcnc: LitexCNC etherbone driver unloaded 
rtapi_app: caught signal 11 - dumping core
free(): invalid pointer
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Waited 3 seconds for master.  giving up.
Note: Using POSIX realtime
motmod: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
Note: Using POSIX realtime
trivkins: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
<commandline>:0: unloadrt failed
Note: Using POSIX realtime

```

Peter-van-Tol, 2023-08-30:
This error is due to an old `loadrt` statement in your hal-files. You have now:
```
loadrt litexcnc
loadrt litexcnc_eth connection_string="192.168.178.150"
```

This should be combined to the following single statement:
```
loadrt litexcnc connection_string="eth:192.168.178.150"
```

Why this error emerges at this moment? It is because the FPGA is reset to its safe state when LinuxCNC is unloaded. This means that `litexcnc` will send a last message to the FPGA. When the FPGA is loaded using two separate statements, the etherbone driver is already unloaded (and memory thus freed up). Thus writing to a closed device, without allocated memory leads to a core dump.

I will close this issue, as the original problem has been solved. In another issue I will unpublish the `litexcnc_eth` component, so it cannot be inadvertently used as a stand-alone component.

Peter-van-Tol, 2023-08-30:
@ozzyrob : for beer that would be then a VB please 🍻 ...

But to be honest: the beer would be on me. Thank you for your support, testing and time spent to make this possible and closing this issue.

closed 2023-08-30 by Peter-van-Tol