Peter-van-Tol/LiteX-CNC #50: Encoder velocity calculation

Peter-van-Tol, 2023-09-24:
### Discussed in https://github.com/Peter-van-Tol/LiteX-CNC/discussions/49

<div type='discussions-op-text'>

<sup>Originally posted by **OJthe123** September 22, 2023</sup>
Hi.
I mentioned this in another question, but it was not the main topic.

I have to multiply the encoder velocity output pin by 8 to get the right value.
Is this a bug, or do I set something wrong?

from my hal....

```
loadrt mult2 names=spindle-rpm-scale
addf spindle-rpm-scale	servo-thread
setp [LITEXCNC](NAME).encoder.00.position-scale 10000
setp [LITEXCNC](NAME).encoder.00.x4-mode 1
setp spindle-rpm-scale.in0 8
net spindle-rpm spindle-rpm-scale.in1 [LITEXCNC](NAME).encoder.00.velocity
net spindle-vel-fb-rps         =>  spindle.0.speed-in	spindle-rpm-scale.out #[LITEXCNC](NAME).encoder.00.velocity

```
when I use the encoder.00.velocity pin directly the speed is wrong</div>

Peter-van-Tol, 2023-09-25:
Sorry about the error. Too tired to spot the error in the buffer => `sizeOf` was giving the size of the array in bytes not the number of items in the buffer. Wasn't thinking clear yesterday evening (got some sleep deprivation from volunteering work). Tested now on my rig and LinuxCNC starts.

OJthe123, 2023-09-25:
Great!
It works on the RPi4 again.
Tomorrow I am going to test it on the lathe 😄

OJthe123, 2023-09-26:
Tested it successfully on my lathe  :+1: 
Encoder velocity now matches with the commanded speed

Peter-van-Tol, 2023-09-26:
Thanks for testing. Will merge it to `main` and will add it to the coming release.

closed 2023-09-27 by Peter-van-Tol