Peter-van-Tol/LiteX-CNC #73: Requirement for encoder.<n>.reset (HAL_BIT)

GTCLive, 2024-01-10:
**Is your feature request related to a problem? Please describe.**
Limitation rather than a problem. I'm working onto an system that requires resetting the counts of an encoder. Since the count pin isn't writable, there isn't a way to reset or zero then encoder. 

**Describe the solution you'd like**
Addition of **<board-name>.encoder.<n>.reset (HAL_BIT)**. I've seen the pin is already there, however it does not work. Would be nice to eventually have the pin functional. 

**Describe alternatives you've considered**
None at this time. If the encoder is moved, the counts are permanent through the lcnc session. Only way for a reset/zeroing at this moment is to power cycle Litecnc and Linuxcnc. Not ideal considering, but that obviously works. 

**Additional context**
I'm happy to examine the code base and compile a custom a version with the encoder.reset functional if you could point me in the correct direction and highlight the implication(s) (if any) of doing so.

GTCLive, 2024-01-10:
Building a MPG with independent X & Z jogging and the linuxcnc component Anglejog on the same MPG. The A axis knob selector enables anglejog, but when I'm changing mode with that knob, the axis rapids to a 'scaled' count of the encoder.counts. 

I've since created an additional set of IN/OUT pins in python that feeds offset counters individually (soft reset?). One for the X & Z, another for Anglejog. This to say my counter reset issue is solved with HAL, bit of a mess though. 

I was asking because I've read it's avail on mesa firmware, and so I thought it was something standard across but yet to be completed in Litexcnc given the pin is already visible. That pin would have cut off a lot of coding and overheads for instance...

Leave it to you to decide whether we need that reset pin. I personally think we do but it's your call man

GTCLive, 2024-01-11:
Closing this since you've read. Open for conversation in the closed area if required, once you've established whether we need count reset, or not! Cheers, Peter, and all the best to you.

Peter-van-Tol, 2024-01-11:
Reopening this as issue, because there is a possible non working HAL pin.

Peter-van-Tol, 2024-01-17:
Proposed changes:

- [ ] add `raw_counts` property, which reflect the number of counts currently held in the register of the FPGA
- [ ] modify `counts` property, which will be reset when the `reset` pin is HIGH in the HAL. Resetting the `counts` property will also affect `position`
- [ ] the `reset` pin will be I/O. After resetting is done, the pin will be set LOW.

This change will not affect any current user. It only adds a functional `reset` pin.

Peter-van-Tol, 2024-01-18:
@GTCLive : would you be able to test this change somewhere next week?

Peter-van-Tol, 2024-01-18:
This feature has been finished. Before merging, if anyone would like to test this, you can try:
``` shell
pip3 install git+https://github.com/Peter-van-Tol/LiteX-CNC@73-requirement-for-encodernreset-hal_bit
```

And run `litexcnc install_driver` to have a working rest-pin.

GTCLive, 2024-01-19:
Peter, I'm honored to be able to test this. Let me se if I can sneak out from home this Saturday, I'll garage myself for a trial. That's awesome btw, and sorry for the delay in responding.

GTCLive, 2024-01-19:
Decided to give it a trial real quick,  but it failed for some strange reason I wish I had the capacity to explain. Attaching the error trace, I'm pretty sure it will make sense to you. 

Restoring the 1.2.0 version and reinstalling its driver makes the machine run normally again. 
$ pip3 install litexcnc==1.2.0
$ sudo -E env PATH=$PATH litexcnc install_driver
 
Awaiting your next command! Good luck. 

**Error log:** [Litexcnc_1.2.1.txt](https://github.com/Peter-van-Tol/LiteX-CNC/files/13993955/Litexcnc_1.2.1.txt)

Peter-van-Tol, 2024-01-19:
Thanks for your quick test. Based on your log I will track the error down. To. Tomorrow evening I will have this sorted out.

Peter-van-Tol, 2024-01-20:
@GTCLive : could you share your config with me? I've trouble to reproduce your error. Maybe it is due to a certain combination of modules. Also, my tests run in `halrun` not in full LinuxCNC. Could you also share your HAL / INI files?

GTCLive, 2024-01-21:
[image]

Much obliged Sir. I attach a stripped down Linuxcnc config & hal which is error-free and work toward @73-requirement-for-encodernreset-hal_bit. The only addon in this config is a simplified mpg to test the encoder0 counter and reset pin. 

However, as you can see from the screengrab, the reset pin will not reset the counter as you've anticipated. Or perhaps I am the wrong person for testing a situation like this one (personally I think I'm the wrong person, particularly when it comes to linux and relevant). 

**GTC_LATHE Config:** [GTC_LATHE.zip](https://github.com/Peter-van-Tol/LiteX-CNC/files/14000353/GTC_LATHE.zip)

Peter-van-Tol, 2024-01-21:
You cannot imagine how much I value your contribution and testing. I will give it another try this evening.

Peter-van-Tol, 2024-01-21:
You were correct: the reset-mechanism was not working. Solved it.

Longer explanation:
- when the `reset` is `HIGH` the current position is stored;
- the position is then subtracted from all subsequent positions received from the FPGA.

One way or the other: I forgot the second step 👎.

If you could test again:
- install with the latest commits: `pip3 install git+https://github.com/Peter-van-Tol/LiteX-CNC@73-requirement-for-encodernreset-hal_bit`
- reinstall the driver: `sudo -E env PATH=$PATH litexcnc install_driver`

It should have reset function now. Also tested it on my machine, with positive results.

GTCLive, 2024-01-22:
Yeah man it works. Wow.
 
However I think I know why you've initially left that reset pin null now (smart). If the counter is reset, in the case of a mpg (for instance), the axis will rapid back to zero unannounced when that count is zerod. The enhancement is adaptable and useful in my case, but for someone else it could become incredibly dangerous. Use/Reset with extreme caution!

That's one of those thing. As previously said; you're the one who calls on whether everybody needs it.

Peter-van-Tol, 2024-01-22:
Whether the axis will rapid back to zero depends in your hal-configuration and the order of the components. I think that I will include the functionality, as the `encoder` module is designed as a drop-in replacement of the LinuxCNC version (which is designed for a LPT-port). This version from LinuxCNC has the same possible danger in it.

_No reset needed for `ANGLEJOG`_
I was not familiar with the function `ANGLEJOG`, so I checked the source code. It allows you to jog to axis simultaneously at a certain angle. While reading the source code I found that you don't have to reset the counts! The code works with a delta between counts of the previous step and the current step.

To make this work without resetting:
- make sure to always update the `counts-in` pin of the `ANGLEJOG` component;
- only enable the component using `enable-in` when the selector is in the correct setting;

This component more or less behaves as a normal jogging command. It internally updates the counter only on the difference between current count and last count when the `counts-in` is TRUE.

_Possible solution for a safe design when you need to reset it for jogging:_
To reset the counter in a safe way, you could use an `AND` between the reset-pin and the jog-enable. If the MPG is being reset, the machine will not be jogged. When the reset is LOW, the `counts` are stable and the jogging will resume. The order of the components / functions is important:
1. Read the FPGA;
2. Determine whether the encoder should be RESET; 
3. Determine jog-enable (AND with RESET)
4. Motion planning (jogging is done here
6. Write to the FPGA

This will result in the following cycle:
1. Read the FPGA -> the encoder is not reset yet, but the axis to be jogged has changed to the rotary axis;
2. Because the axis has changed, the `counts` of the encoder should be reset
3. Because the reset is active, jogging is prohibited 
4. Motion planning: won't move as jogging is not enabled
5. Write to the FPGA
6. Read the FPGA -> the encoder `counts` value is set to 0 now;
7. The axis didn't change, so the RESET will be low
8. Jogging is enabled
9. Motion planning will jog the relative rotation
10. Write to FPGA

GTCLive, 2024-01-23:
That's beyond awesomeness of your part for having taken the time to look into this, Peter. Thoughtful thinking in where and how to use the reset pin too. Much appreciated, I'll revise the work I've done and see how I can make it better.

On the testing part I can say that it works at my end. Do let me know if you need me to carry out other tests I'll be happy to give this a go, despite limited abilities. (ps. your knowledge is incredible btw!)

GTCLive, 2024-01-24:
You were right - as always. As long as you feed the count-in it won't complain nor shoot unannounced.  I'm including the final HAL code for future reference in hope that it helps someone else.  Note the 'static' 60.0degree angle, that's just for testing but can be placed in a textbox in a py UI anywhere. Works great - replicates a manual lathe although entirely CNCd. Crazy stuff!! 

`
### 2AX LATHE MPG (X & Z) with ANGLEJOG enabled by the A Axis Selector.

loadrt anglejog names=anglejog
addf   anglejog servo-thread

loadrt mux4 names=mpg-scale
addf mpg-scale servo-thread

loadrt mux8 names=mpg-axis
addf mpg-axis servo-thread

loadrt or2 count=4
addf or2.0 servo-thread
addf or2.1 servo-thread
addf or2.2 servo-thread
addf or2.3 servo-thread

loadrt ilowpass
addf ilowpass.0 servo-thread
setp ilowpass.0.gain 0.02
setp ilowpass.0.scale 1
setp mpg-scale.in0 0.0
setp mpg-scale.in1 0.001
setp mpg-scale.in2 0.010
setp mpg-scale.in3 0.050

### COUNTER
setp [CARD](NAME).encoder.00.x4-mode 0
net jog-smoothing ilowpass.0.in  <= [CARD](NAME).encoder.00.counts
net jog-position <= ilowpass.0.out
net jog-position => axis.x.jog-counts => joint.0.jog-counts
net jog-position => axis.z.jog-counts => joint.1.jog-counts	

### SCALE SELECTORS
net scale1 or2.0.in0 	<= [CARD](NAME).gpio.05.in-not
net scale2 or2.1.in0 	<= [CARD](NAME).gpio.06.in-not
net scale3 or2.0.in1 or2.1.in1 <= [CARD](NAME).gpio.07.in-not
net pos2 mpg-scale.sel0 <= or2.0.out
net pos3 mpg-scale.sel1 <= or2.1.out
net axis-mpg-jog-scale 	axis.x.jog-scale  joint.0.jog-scale <= mpg-scale.out
net axis-mpg-jog-scale 	axis.z.jog-scale  joint.1.jog-scale <= mpg-scale.out

### AXIS SELECTOR
net x-jog-select => or2.2.in0 <= [CARD](NAME).gpio.00.in-not
net z-jog-select => or2.3.in0 <= [CARD](NAME).gpio.02.in-not
net a-jog-select => or2.2.in1 => or2.3.in1 <= [CARD](NAME).gpio.03.in-not
net axis_x mpg-axis.sel0 => axis.x.jog-enable <= or2.2.out
net axis_z mpg-axis.sel1 => axis.z.jog-enable <= or2.3.out

### ANGLEJOG (https://linuxcnc.org/docs/devel/html/man/man9/anglejog.9.html)
###--------------------------------------------------------------

setp anglejog.angle-degrees-in		60.0	# 60degree (***this needs to be placed into the PyVCP UI***)

setp anglejog.max-vel				10.0	# 10% of the joint velocity
setp anglejog.max-accel				150.0	# Set to joint max accel
setp anglejog.accel-fraction-in		10.0
net jog-position 					=> anglejog.counts-in
net a-jog-select 					=> anglejog.enable-in
net axis-mpg-jog-scale 				=> anglejog.scale-in
net ajog_x 						<= anglejog.coscounts
net ajog_xfraction 					<= anglejog.cos-accel-fraction
net ajog_xfraction 					=> axis.x.jog-accel-fraction
net ajog_z						<= anglejog.sincounts
net ajog_zfraction					<= anglejog.sin-accel-fraction
net ajog_zfraction					=> axis.z.jog-accel-fraction

`

GTCLive, 2024-01-24:
You were right - as always. As long as you feed the count-in it won't complain nor shoot unannounced.  I'm including the final HAL code for future reference in hope that it helps someone else.  Note the 'static' 60.0degree angle, that's just for testing but can be placed in a textbox in a py UI anywhere. Works great - replicates a manual lathe although entirely CNCd. Crazy stuff!! 

`
###2AX LATHE MPG (X & Z) with ANGLEJOG enabled by the A Axis Selector.

loadrt anglejog names=anglejog
addf   anglejog servo-thread

loadrt mux4 names=mpg-scale
addf mpg-scale servo-thread

loadrt mux8 names=mpg-axis
addf mpg-axis servo-thread

loadrt or2 count=4
addf or2.0 servo-thread
addf or2.1 servo-thread
addf or2.2 servo-thread
addf or2.3 servo-thread

loadrt ilowpass
addf ilowpass.0 servo-thread
setp ilowpass.0.gain 0.02
setp ilowpass.0.scale 1
setp mpg-scale.in0 0.0
setp mpg-scale.in1 0.001
setp mpg-scale.in2 0.010
setp mpg-scale.in3 0.050

###COUNTER
setp [CARD](NAME).encoder.00.x4-mode 0
net jog-smoothing ilowpass.0.in  <= [CARD](NAME).encoder.00.counts
net jog-position <= ilowpass.0.out
net jog-position => axis.x.jog-counts => joint.0.jog-counts
net jog-position => axis.z.jog-counts => joint.1.jog-counts	

###SCALE SELECTOR
net scale1 or2.0.in0 	<= [CARD](NAME).gpio.05.in-not
net scale2 or2.1.in0 	<= [CARD](NAME).gpio.06.in-not
net scale3 or2.0.in1 or2.1.in1 <= [CARD](NAME).gpio.07.in-not
net pos2 mpg-scale.sel0 <= or2.0.out
net pos3 mpg-scale.sel1 <= or2.1.out
net axis-mpg-jog-scale 	axis.x.jog-scale  joint.0.jog-scale <= mpg-scale.out
net axis-mpg-jog-scale 	axis.z.jog-scale  joint.1.jog-scale <= mpg-scale.out

###AXIS SELECTOR
net x-jog-select => or2.2.in0 <= [CARD](NAME).gpio.00.in-not
net z-jog-select => or2.3.in0 <= [CARD](NAME).gpio.02.in-not
net a-jog-select => or2.2.in1 => or2.3.in1 <= [CARD](NAME).gpio.03.in-not
net axis_x mpg-axis.sel0 => axis.x.jog-enable <= or2.2.out
net axis_z mpg-axis.sel1 => axis.z.jog-enable <= or2.3.out

###ANGLEJOG (https://linuxcnc.org/docs/devel/html/man/man9/anglejog.9.html)

setp anglejog.angle-degrees-in		60.0	# 60degree (***this needs to be placed into the PyVCP UI***)

setp anglejog.max-vel				10.0	# 10% of the joint velocity
setp anglejog.max-accel				150.0	# Set to joint max accel
setp anglejog.accel-fraction-in		10.0
net jog-position 					=> anglejog.counts-in
net a-jog-select 					=> anglejog.enable-in
net axis-mpg-jog-scale 				=> anglejog.scale-in
net ajog_x 						<= anglejog.coscounts
net ajog_xfraction 					<= anglejog.cos-accel-fraction
net ajog_xfraction 					=> axis.x.jog-accel-fraction
net ajog_z						<= anglejog.sincounts
net ajog_zfraction					<= anglejog.sin-accel-fraction
net ajog_zfraction					=> axis.z.jog-accel-fraction
`

GTCLive, 2024-01-24:
Save me the embarrassment and Close This. Still good to have that once errant reset pin finally working nonetheless, sure will come handy if needed for everything else. Thank you, Peter.

closed 2024-01-24 by Peter-van-Tol