Peter-van-Tol/LiteX-CNC #86: New module `shift_in` and `shift_out` for using shift registers

Peter-van-Tol, 2024-02-19:
### Discussed in https://github.com/Peter-van-Tol/LiteX-CNC/discussions/52

<div type='discussions-op-text'>

<sup>Originally posted by **Peter-van-Tol** October 15, 2023</sup>
When using the 5A-75B, one can get at the point that 51 pins are not enough. The Arduino platform offers two libraries for shifting data using buffers:
- 74HCT595 for output;
- 74HC165 for input;

Using these chip, one can use for example a pendant with loads of functions, connecting the MPG directly to the pins and connecting the buttons to these shift registers.

Design spec:
- each module can handle up to 32 output (`shift_out`) or 32 inputs (`shift_in`);
- pins required are `DATA`, `CLK` and `LATCH`;
- input and output cannot be mixed on one channel;
- support for multiple channels, no overlap in pins. 

Should this prove useful?</div>

Peter-van-Tol, 2025-03-29:
The work on this has been continued in a different branch: #127 . This is due to some structural changes in the setup of the firmware. 

The `shift-out` module has been finished and tested. It has the following features:
- based on `74HCT595` chipset and timings; 
- the pre-scaler if currently fixed at 1000 (i.e.: when the the FPGA runs at 50 MHz, the `74HCT595` sees 50 kHz). This pre-scaler should be improved so it scales to a fixed frequency of 100 kHz instead of a fixed pre-scaler.
- supports up to 64 extra GPIO (in total 8 chips can be chained together). This is based on a 1 kHz loop in LinuxCNC, combined with a fixed frequency of 100 kHz;

A board has been developed for the 74HCT595. The v1.0 version of this board is confirmed to be working. The shift-out board has been designed to work together with the GPIO out board. However, the v1.0 of the latter did not work.

[image]

The configuration of this new-module is relative simple:
``` json
    "modules": [
        {
            "module_type": "gpio_expander",
            "instances": [
                {
                    "serial_data_pin": "j3:2",
                    "clock_pin": "j3:3",
                    "latch_pin": "j3:4",
                    "clear_pin": "j3:5",
                    "ports": [
                        {}
                    ]

                }
            ]
        }
    ]
```

This configuration has 1 GPIO port expander, using the 74HCT595 (it is inferred based on the pin names). The pins on this port do not have aliases. When you load the board with this configuration, the following pins are created in the HAL.
```
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.00.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.01.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.02.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.03.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.04.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.05.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.06.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.0.07.out
```

Things to improve on the design of this new module:
- [ ] generating aliases for the GPIO expander (i.e. `EMCO5_SPI.gpio.0.00.out` -> `EMCO5_SPI.gpio.LED_ON`). The aliases share the namespace of the regular IO.
- [ ] improve the configuration by adding the used chipset (i.e. 74HCT595). Not strictly required, but it makes the configuration more readable
- [ ] support more then one chain (index is hard-coded, should be easy to fix).

At this moment I'm waiting for the shift-in board to arrive. When that board is received, this driver can be further extended.

Peter-van-Tol, 2025-03-30:
The work on this has been continued in a different branch: #127

Peter-van-Tol, 2025-03-30:
Updated the configuration for readability (snippet):
``` json
            "module_type": "gpio_expander",
            "instances": [
                {   
                    "expander_type": "shift_out",
                    "pins": {
                        "serial_data": "j3:2",
                        "clock": "j3:3",
                        "latch": "j3:4",
                        "clear": "j3:5"
                    },
                    "ports": [
                        {}
                    ]
                }
            ]
```

Peter-van-Tol, 2025-03-30:
Fixed the support of multiple chains:
```
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.00.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.01.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.02.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.03.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.04.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.05.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.06.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.07.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.00.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.01.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.02.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.03.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.04.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.05.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.06.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.01.07.out
```

Peter-van-Tol, 2025-03-30:
Also implemented aliases for the port expander:
```
        {
            "module_type": "gpio_expander",
            "instances": [
                {   
                    "expander_type": "shift_out",
                    "pins": {
                        "serial_data": "j3:2",
                        "clock": "j3:3",
                        "latch": "j3:4",
                        "clear": "j3:5"
                    },
                    "ports": [
                        {
                            "0": "start-LED",
                            "4": "pause-LED"
                        }
                    ]
                }
            ]
        }
```

The above wil result in:
```
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.01.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.02.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.03.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.05.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.06.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.00.07.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.pause-LED.out
     4  bit   IN          FALSE  EMCO5_SPI.gpio.start-LED.out
```

Peter-van-Tol, 2025-03-30:
The board for the `shift_in` is expected to arrive coming week, then I will finish this module.

horcsab, 2025-06-06:
Hello Peter,