Peter-van-Tol/LiteX-CNC #87: Maximum frequency 375 kHz.

lex27087, 2024-02-20:
Dear friend, you have done a great job.
Thank you for that.
I would really like to know why it is possible to generate the maximum 375 kHz?
And this is at a FPga frequency of 50 MHz.
I tried to set the maximum frequency to 55 MHz - 200 kHz.
60 MHz - 200 kHz
70MHz -200KHz
80MHz - 306 kHz
100 MHz - 200 kHz.
I would really like to know what the limitation is?
Thank you.
[image]

lex27087, 2024-02-20:
[image]

Peter-van-Tol, 2024-02-20:
To be sure: you are talking about the `PWM` module?

lex27087, 2024-02-20:
tried 200ns
steplen
200ns
step-space
[image]

40Mhz

lex27087, 2024-02-20:
at 40 MHz
more than 306 kHz
does not work.
staplen 200ns

lex27087, 2024-02-20:
Is it possible to remove the frequency limit?

romanetz, 2024-02-20:
Please clarify exactly what does not work.
From our conversation in a local telegram group I assume that you cannot
set maximum velocity over than that produces specified output frequency
because it causes "following error"

lex27087, 2024-02-20:
[image]

Peter-van-Tol, 2024-02-20:
Just to acknowledge: the combination of `steplen` and `stepspace` should be able to give you 500 kHz, which is limited in the software to 400 kHz. I will check whether the scaling takes a value above 400 kHz or the first value below 400 kHz. Should be set to the value just above 400 kHz. 

You have also found a bug: the maximum frequency of the FPGA is not taken into account in the driver. So it is possible to push the FPGA harder then it can count from LinuxCNC. This will cause all kind of issues on its own and thus lead to a possible Joint following error.

lex27087, 2024-02-20:
Thank you very much)
I would like to get higher frequencies.
For controlling servomotors.

lex27087, 2024-02-20:
@Peter-van-Tol 
if your equipment cannot see above frequency.
You can send me corrections and edits.
I'll give you feedback quickly.
Thank you

Peter-van-Tol, 2024-02-20:
When using 40 MHz, overflow in the speed register occurs. Sending a value of 0 to the FPGA. 

The FPGA uses fixed point math. To determine the location of the decimal point, a value `shift` is calculated. In this calculation, I forgot that the data range for a _signed_ integer is smaller. 

Which version of LitexCNC are you using? In the current version (`1.2.4`) it actually prohibits you from setting the frequency this high (also a bug, but the effects cancel each other out).

lex27087, 2024-02-20:
@Peter-van-Tol 
1.2.4

Peter-van-Tol, 2024-02-20:
Changing this bug will lead to a required recompilation of the firmware. Otherwise re-installing the driver will cause the stepper to move too slow with respect to the commanded speed if user do update the driver, but not their firmware. Therefore, changing this will be done in the upcoming version 1.3.0. This release is estimated to be available beginning of March. 

As an intermediate solution I'm going to cap the frequency on the driver side. Such an error may lead to dangerous situations, how unlikely they are to happen. Capping the frequency is driver side only, so can be safely done. I'll can test this on my own setup.

Proposed solution:
- make maximum frequency a parameter of the config. Users can make a trade-off between resolution and speed.
- this parameter is locked in the firmware and is read during the config-phase.
- the calculation of the `shift` is removed from the driver.

lex27087, 2024-02-20:
compilation is not a problem.
are we waiting for 1.3.0?
I didn’t understand a little what to do.
Thank you

Peter-van-Tol, 2024-02-20:
Will make a new version you can test. As soon as finished, I give instructions on how to install. When testing is OK, the branch will be merged and released as 1.3.0. With some luck, you can test tomorrow.

lex27087, 2024-02-20:
Thank you very much.

Peter-van-Tol, 2024-02-20:
[image]
 
The maximum frequency is now correctly reported in the HAL. The maximum frequency is determined based on the config on the FPGA. 

Could you please test whether you can achieve higher speeds with this modification?

To install this version:
```bash
pip install git+https://github.com/Peter-van-Tol/LiteX-CNC/issues/87@87-maximum-frequency-375-khz
sudo env "PATH=$PATH" litexcnc install_driver
```

You have to recompile your firmware and upload it to the card.

lex27087, 2024-02-20:
@Peter-van-Tol 
Okay, we'll check it out a little later.
Thank you

lex27087, 2024-02-20:
@Peter-van-Tol 
Do I understand you correctly?
Will the generation frequency depend on the configuration file?
40 MHz
50 MHz
What is the maximum FPGA frequency you can set?

Peter-van-Tol, 2024-02-20:
@lex27087:

In the config there are two parameters:
- `clockfrequency`: this is the frequency of the FPGA. If set too high, timing constraints won't be met. Normally a value between 30 and 50 Mhz is good.
- `max_frequency`: a new parameter set on the stepgen instances. Default value is 400 kHz. This max frequency is the minimum frequency the stepgen can reach.

The actual frequency depends on the FPGA clock and the scaling. For example: when using a 50 Mhz clock and 400 kHz max frequency, the scale factor is 2^5 and the actual maximum frequency is 790 kHz.

If you want to obtain a specific frequency, you should set the `max_frequency` parameter in the JSON configuration. So feel free to put it at 750 kHz if you want to. Do not alter the clock of the FPGA, it is not designed for that.

I will add a snippet with an example configuration later.

lex27087, 2024-02-20:
@Peter-van-Tol 
Everything is fine.
Let's check it out now.
Thank you for answering so quickly.

Peter-van-Tol, 2024-02-20:
Just committed some more code:
- timings are now indvidual per stepgen. Making it easy to mix different drivers within a machine
- the maximum frequency of a stepgen channel, as reported in the HAL, is now also dependent on the `steplen` and `stepspace`
- warning when stepping too fast is reinstated

The changed module does compile and run my machine. @lex27087 : could you do some more high speed testing. If the system is working, then I will merge this into `main`.

Peter-van-Tol, 2024-02-20:
An example of a stepgen which supports 800 kHz stepping:
```json
{
            "module_type": "stepgen",
            "instances": [
                {
                    "pins" : {
                        "stepgen_type": "step_dir",
                        "step_pin": "j1:1",
                        "dir_pin": "j1:2"
                    },
                    "max_frequency": 800000
                    "soft_stop": true
                },
...
}
```

lex27087, 2024-02-21:
@Peter-van-Tol 
I'll definitely check it in a few hours.
It's still late at night.
And I'll check it on an oscilloscope.
We managed to generate 774 kHz.
We haven’t gone any further yet.
When I'm at work, I'll definitely check it at frequencies from 40 to 60.
Thank you

lex27087, 2024-02-21:
@Peter-van-Tol 
[image]

[image]

[image]

[image]

[image]

lex27087, 2024-02-21:
@Peter-van-Tol 
along the signal edges.
here it is not connected directly to fpga.
That's why I think there are distortions.

lex27087, 2024-02-21:
@Peter-van-Tol 
[image]

Peter-van-Tol, 2024-02-21:
I think that the modification for the max frequency has been succesful. Thanks for testing. The edges are definitely due to the electric system. Reminds me of my old BOB called `1208`. The opto-couplers were so damn slow, you could barely reach 10 kHz step frequency before the signal got lost. Running at 17 MHz step frequency is, let say it honest, insane. What frequency you are going to drive your servos?

@hmnijp : still some work to do to get the individual timings correct. Will look into that!

To answer your question: the clock frequency is always a multiple of the max frequency, because the max frequency is calculated with:
```
max_frequency = clock_frequency / 2**(shift+1)
```

The shift is determined so overflow of the velocity command, as sent to the FPGA, will not overflow. With 40 MHz clock frequency and a desired 400 kHz step frequency the `shift` equals 5. The leads to a theoretical maximum attainable step frequency  of 625 kHz. Dividing it once more with a factor 2 would lead to a maximum step frequency then desired 400 kHz.

The practical step frequency is also limited by the `steplen` and `stepspace` parameters (both parameters in ns):
```
max_frequency = 1 / ((stepspace + steplen) * 10e-9)
```

My setup uses a drive with 5000 ns timing for both `stepspace` and `steplen`. This results in a 100 kHz stepping frequency for my stepper (`DM422`). To optimize the velocity resolution, I can opt to change the `max_frequency` in my config to 100000 Hz for an optimal system.

lex27087, 2024-02-21:
@Peter-van-Tol 
Well, the 1.5 m wire most likely plays a role here.
frequencies are high.
For me this is not critical.
500 kHz is enough for me.
thanks for the work you've done.
We are waiting for minor fixes and a new driver 1.3.0

Peter-van-Tol, 2024-02-21:
I just added a commit to this branch which solves the error of the broken stepgen when individual timings are applied. The stepper here was nicely spinning....

Peter-van-Tol, 2024-02-22:
Thank you. Back to the drawing board it is

Peter-van-Tol, 2025-01-09:
Has been resolved in version 1.3.0, which has been released just now. The problems have been resolved in different issues:
- not all stepgen responding #113, #79. This was due to a mismatch in memory width 
- determine shift #124. The shift is calculated in firmware and stored in CSR register

closed 2025-01-09 by Peter-van-Tol