Peter-van-Tol/LiteX-CNC #114: [Feature request]:

romanetz, 2024-10-22:
### Initial Checks

- [X] I have searched Google & GitHub for similar requests and couldn't find anything
- [X] I have read the documentation and couldn't find anything

### Description

I started to investigate a possibility to integrate vexriscv processor core into the firmware. This allows to add modbus RTU support over RS-485, SPI master to interact with external ADCs and DACs, provide a kind of autonomous functionality such as electronic cam and support of modbus TCP to ease integrations with industrial process automation. 

### Affected Components

- [X] CLI.
- [X] Supported boards (i.e. 5A-75B, 5A-75E, HUB75HAT, etc.)
- [X] Supported connections (i.e. SPI, Ethernet, etc.)
- [X] Modules (i.e. GPIO, PWM, stepgen, encoders, etc.)
- [X] Documentation

Peter-van-Tol, 2024-10-22:
Do you have any references to a specific board that utilizes this processor?

Tonadd a new board, the following steps are required:
- add board name to pyproject.toml (otherwise the compiler cannot find it)
- add the base firmware to the source code. This can be very similar to the code from the litex package.
- if the firmware cannot be compiled by Yosys (such as with Zylinx and Gowin) additional steps are required either in the source code or bybthe user using this board. For example, with Zylinx you can generate the code, but it has to be compiled in their software.

romanetz, 2024-10-23:
As a first step, I added i9 board into litexcnc python sources and built a new firmware for the board. Some changes in configurations were done: I changed pin names like "jx:y" to "pmodx:y" as stated in platform definition. Next step will be to add vexriscv cpu into the configuration.