Peter-van-Tol/LiteX-CNC #117: [Feature request]: Support user LED and button

Peter-van-Tol, 2024-11-01:
### Initial Checks

- [X] I have searched Google & GitHub for similar requests and couldn't find anything
- [X] I have read the documentation and couldn't find anything

### Description

Support the user led and buttons on the board. These are actually on the `io`, and not on the connectors, so they can not be mixed. Steps to take:

- [ ] Convert the user led and button to a connector (I know, they are not a connector, but at least they can be used in that case)
- [ ] Ensure the user LED can only be used as output (supported in PWM and GPIO)
- [ ] Ensure the user Button can only be used as input (supported in GPIO)

Misconfiguring the FPGA should not lead to damage to the board.

### Affected Components

- [ ] CLI.
- [X] Supported boards (i.e. 5A-75B, 5A-75E, HUB75HAT, etc.)
- [ ] Supported connections (i.e. SPI, Ethernet, etc.)
- [X] Modules (i.e. GPIO, PWM, stepgen, encoders, etc.)
- [ ] Documentation

chaotix-, 2024-11-02:
Thank you for your work. I just tested this branch and the LED as well as the button work as GPIO pins.

I also noticed a couple of problems:
1. The LED GPIO-pin does not have `invert-output` only `out`.
2. Adding `"name":"something"` to the JSON is ignored. No aliases for the LED or button pin are added to `alias.hal`
3. When a PWM-section is present in the JSON file building the firmware throws an error. It's not necessary to reference user_led oder user_btn to get this error.
```
$ litexcnc build_firmware cncpi-5a-75e_v8.2-userled.json --build -a
Traceback (most recent call last):
  File "/home/lars/litexcnc/bin/litexcnc", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/cli/build_firmware.py", line 37, in cli
    firmware_config = LitexCNC_Firmware.parse_raw(' '.join(config.readlines()))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/pydantic/main.py", line 549, in parse_raw
    return cls.parse_obj(obj)
           ^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/pydantic/main.py", line 526, in parse_obj
    return cls(**obj)
           ^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/soc.py", line 65, in __init__
    current_module = subclass(**current_module) 
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/pydantic/main.py", line 341, in __init__
    raise validation_error
pydantic.error_wrappers.ValidationError: 4 validation errors for PWM_ModuleConfig
instances -> 0 -> pins
  field required (type=value_error.missing)
instances -> 1 -> pins
  field required (type=value_error.missing)
instances -> 2 -> pins
  field required (type=value_error.missing)
```

This is the PWM section that produced the error:
``` 
       }, {
            "module_type": "pwm",
            "instances": [
                {"pin":"j7:1", "name":"pwm1"},
                {"pin":"j7:5", "name":"pwm2"},
                {"pin":"j7:6", "name":"pwm3"}
            ]
        }, {
```

As soon as I delete this section the firmware compiles just fine.

Peter-van-Tol, 2024-11-02:
Thank you for testing!

Concerning your remarks:
1. The `invert_output` is a HAL parameter, not HAL pin. In `halrun` you can show the available parameters with `show param`
2. Can you confirm you've build the firmware with the `-a` tag enabled? Only with this tag the `alias.hal` will be created.
3. The `main` branch has more upcoming changes in it. Those changes also ended up in this developing branch. The reason why it does not compile is that PWM now support different modes. Replace the PWM section of your config with below to get it working:
   ``` json
        {
            "module_type": "pwm",
            "instances": [
                {
                    "pins": {
                        "output_type": "updown",
                        "up": "j2:1",
                        "down": "j2:2"
                    }
                }, {
                    "pins": {
                        "output_type": "pwmdirection",
                        "pwm": "j2:4",
                        "direction": "j2:5"
                    }
                }, {
                    "pins": {
                        "output_type": "single",
                        "pwm": "j2:6"
                    }
                }, {
                    "pins": {
                        "output_type": "single",
                        "pwm": "j2:7"
                    }
                }
            ]
        }
   ```

chaotix-, 2024-11-02:
1. Oh. My bad.
2. Sorry. Also my fault. I built the firmware with `-a` but in my json file the name attribute was wrong. `"name:":"user_led"` which, of course, does not work. When removing the additional ':' I get an alias for the GPIO pin.
3. Thank you. This compiles and works beautifully. Now I have a nice pulsating onboard LED as soon as I start LinuxCNC and when pulling the network plug the brightness jumps to 100% instantly (at least instantly for my human eyes) and continues pulsating as soon as I plug it back in.

But I still don't get any aliases for the PWM pins. I tried the following:
```
{
    "module_type": "pwm",
    "instances": [
	{
	    "pins": {"output_type": "single", "pwm":"user_led:1", "safe_state":true, "name":"user_led"}
        }
    ]
}
```

and I also tried:
```
{
    "module_type": "pwm",
    "instances": [
	{
	    "pins": {"output_type": "single", "pwm":"user_led:1"},
            "safe_state":true,
	    "name": "user_led"
        }
    ]
}
```
Both variants don't generate a alias for the PWM-pin in `alias.hal`.

`"safe_state":true` is also not working as expected. You wrote the LED is active low so the watchdog should turn off the LED, but it sets it to 100% brightness. But maybe both config file variants are wrong and I need to specify `"name"` and `"safe_state"` differently.

Peter-van-Tol, 2024-11-02:
The `safe_state` is only available on GPIO and not on PWM. Good call to also add it to those pins to set their state before LinuxCNC kicks in. I will also look into the alias of the PWM.

chaotix-, 2024-11-03:
Do you have a correct PWM section that compiles? Using both variants I tried above I get the following error on trying to build the firmware:

```
Traceback (most recent call last):
  File "/home/lars/litexcnc/bin/litexcnc", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/cli/build_firmware.py", line 50, in cli
    soc = firmware_config.generate()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/soc.py", line 120, in generate
    module.create_from_config(soc, watchdog)
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/config/modules/pwm.py", line 216, in create_from_config
    PwmPdmModule.create_from_config(soc, watchdog, self)
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/modules/pwm.py", line 175, in create_from_config
    soc.platform.add_extension([
                               ^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/modules/pwm.py", line 176, in <listcomp>
    ("pwm", index, *pwm_instance.pins.convert_to_signal())
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/config/modules/pwm.py", line 46, in convert_to_signal
    Subsignal("pwm", Pins(self.pwm), IOStandard(self.io_standard)),
                     ^^^^^^^^^^^^^^
  File "/home/lars/toolchain/litex/litex/litex/build/generic_platform.py", line 36, in __init__
    self.identifiers += i.split()
                        ^^^^^^^
AttributeError: 'NoneType' object has no attribute 'split'
```

When I delete the PWM config block in the JSON file the firmware builds without a problem. I tried the `main` and the `119-feature-request-revamp-watchdog` branches.

Peter-van-Tol, 2024-11-03:
On the list for tonight.

closed 2024-11-03 by Peter-van-Tol

chaotix-, 2024-11-04:
Works great. Thank you.

PR #118 (2024-11-02): 117 feature request support user led and button
Completed and tested

closes #117