Peter-van-Tol/LiteX-CNC #119: [Feature request]: Revamp Watchdog

Peter-van-Tol, 2024-11-02:
### Initial Checks

- [X] I have searched Google & GitHub for similar requests and couldn't find anything
- [X] I have read the documentation and couldn't find anything

### Description

Revamp the watchdog:
- add functions:
  - enable -> this pin will be HIGH as long the watchdog is happy;
  - heartbeat -> this pin will show a heartbeat as long as the watchdog is happy;
  - charge_pump -> configurable charge pump for external components;
  - optionally -> tie in the E-stop so the E-stop circuit is completely onboard. 

Also, the driver has to be checked. Pulling the cable and then plugging it in again, will resume movement. The communication should be restored, but the watchdog should be actively reset before any action/movement is commenced.

### Affected Components

- [ ] CLI.
- [ ] Supported boards (i.e. 5A-75B, 5A-75E, HUB75HAT, etc.)
- [ ] Supported connections (i.e. SPI, Ethernet, etc.)
- [X] Modules (i.e. GPIO, PWM, stepgen, encoders, etc.)
- [ ] Documentation

Peter-van-Tol, 2024-11-02:
Status: added enable and heartbeat function. Example config for HUB75HAT:
``` json
    "watchdog": {
        "functions": [
            {
                "function_type": "enable",
                "pin": "ena:0"
            },
            {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true
            }
        ]
    },
```

The heartbeat is really pulsating ❤️

Peter-van-Tol, 2024-11-03:
For the HUB75HAT I found the following configuration very promising:

``` json
    "watchdog": {
        "functions": [
            {
                "function_type": "enable",
                "pin": "ena:0"
            },
            {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true,
                "waveform": "heartbeat",
                "speed": 0.857
            }
        ]
    },
```

chaotix-, 2024-11-04:
I have not tested this branch yet, but I have a question regarding the watchdog:

How do you reset it? The documentation says that you should write `0` to `<board-name>.watchdog.has_bitten`. But how do you do that when you have already connected the `has_bitten`-pin to a HAL net for the estop? I cannot even do it manually via halshow as it complains that it is already connected to a net.

Do you have a working example that triggers the estop and is able to be reset? Or should the watchdog have a `reset`-HAL-pin?

Peter-van-Tol, 2024-11-04:
The issue you're bringing up now is exactly the reason why I want to revamp the watchdog. 

In the previous design, the enable output of the watchdog was HIGH when the watchdog didn't bite. When the card was powered on, but yet not controlled by LinuxCNC (i.e. the stream of messages was not started), the enable was HIGH as well. This is not safe in my opinion, so this has changed in the current version. When the card is started, the ENA stays low until the watchdog is activated from LinuxCNC.

As long as LinuxCNC is running and sending messages, the time-out is being updated; the watchdog is kept happy. When the messages stop from LinuxCNC (for example by pulling the Ethernet-cable), the watchdog bites and the ENA is pulled low again and the heartbeat stops. Up to this point I'm happy with the behaviour of the watchdog.

What happens now is that the watchdog _resumes_ as soon the connection is restored. This is not correct yet. The design should be:
- the watchdog bites;
- the status of the watchdog is being read by the system (the bug lies most-likely within this step);
- when the watchdog has bitten, the driver will prevent update of the timer, thus the state is stable;
- the HAL output `has_bitten` can be used within the E-stop routine to halt LinuxCNC;
- the watchdog can be reset by setting the HAL pin `has_bitten` to low, which resumes updates (current design, can be modified to suit the general needs).

Challenge is that I use SPI for testing the board, so I cannot pull the cable. I use `halrun` for testing purposes, so I can use the commands `start` and `stop` to start and stop the communication. However, when restarting the communication I don't see the status that the watchdog has bitten. 

Do you have a preference for a method to reset the watchdog?

chaotix-, 2024-11-04:
I think the problem with the approach is that `has_bitten` is used as input and output, but you can only connect it to one net, so at least I don't know how to trip the estop in HAL and reset the watchdog. I would use `estop_latch` as inspiration and just use an extra `reset` input pin to reset the watchdog and `has_bitten` is used just as output pin.

Did you try `sudo dtparam spi=off` to stop the SPI interface and maybe trigger the watchdog? I don't know if it works when the SPI interface is in use. `sudo dtparam spi=on` should resume SPI.

Peter-van-Tol, 2024-11-04:
Using `start` and `stop` in `halrun` does indeed trigger the `watchdog`. The problem of the watchdog being reset automatically is due to timing issues in the FPGA. The enable signal going LOW arrived one clock-cycle too late, which sparked a new cycle of the timeout being replenished. 

I have changed the firmware so the timing paths of both `enable` and `timeout` have equal length. Now, when the watchdog bites, it will keep on biting when the communication is restored. Also the `has_bitten` pin remains HIGH, so it can be used as input for the software EStop.

At this moment the only way to reset the `watchdog` is by restarting the software (no power cycle required, the FPGA is reset to a predefined state at start-up). The method of driving `has_bitten` low does not work as intended. I will implement your suggested solution by getting inspiration from the `estop_latch`. Thanks for this advice.

PS: in this branch PWM is not working. It was branched off before the PWM was fixed in `main`.

Peter-van-Tol, 2024-11-07:
Added the pin `reset` to the watchdog. On a rising edge, this will reset the watchdog and the FPGA will resume operation.

To include this version of the watchdog into LinuxCNC, you can add the following lines to the HAL:
```
net estop-loopout iocontrol.0.emc-enable-in <= EMCO5_SPI.watchdog.has_bitten
net estop-reset iocontrol.0.user-request-enable => EMCO5_SPI.watchdog.reset
```

The above version will trigger an E-Stop in LinuxCNC when communication is stopped. When the user requests to enable the machine, the reset will be pulled high and the watchdog will resume.

Above example does have its limitations at this moment:
- the Watchdog does not respond to an E-Stop coming from LinuxCNC;
- the E-Stop connected to the FPGA have to travel through the LinuxCNC;
- the standard `estop_latch` can not be used because of the delay between sending the reset signal and receiving the OK status (1 period, while `estop_latch` expects direct response.

I want to integrate the E-Stop in the watchdog. Buttons and alarms will directly trigger the watchdog, which will inform LinuxCNC that a E-Stop has happened _and_ which E-stop has been triggered. Also, the watchdog will get a `fault-in` pin, so any  E-stop comin from LinuxCNC can also the watchdog. The watchdog will thus also act as latch!

For `estop_latch` I will write a new component in LinuxCNC, so it allows for a configurable delay.

chaotix-, 2024-11-08:
I think you can get `estop_latch` to work with you existing approach. There is a `timedelay` component (http://linuxcnc.org/docs/html/man/man9/timedelay.9.html) so you can write to `watchdog.reset` a couple of milliseconds before you write to `estop.N.reset`.

If you want you can push your changes to this branch so I can test if I can get this to work.

Peter-van-Tol, 2024-11-08:
You're correct, there is no need for an new component.

I've implemented an `estop_latch` within the watchdog. I've added the following pins to the watchdog:
- watchdog.ok-in bit in (default: true);
- watchdog.fault-in bit in (default: false);
- watchdog.reset bit in;
- watchdog.ok-out bit out (default: false);
- watchdog.fault-out bit out (default: true);

The initial state is "Faulted". When faulted, the `out-ok` output is `false`, the `fault-out` output is `true`. The heartbeat on the watchdog will beat, indicating that communication is OK. The enable function will be LOW when the `fault-out` is `true.

The reset command to the watchdog is send when all these conditions are true:
- `fault-in` is `false`;
- `ok-in` is `true`;
- `reset` changes from `false` to `true`.

On the watchdog additional checks will be performed (not implemented yet, but think off E-Stop and ALARMS tied into the wathdog).

When "OK", the `out-ok` output is `true`, the `fault-out` output is `false`. The enable function will be HIGH.

The state changes from "OK" to "Faulted" when any of the following are true:
- `fault-in` is `true`;
- `ok-in` is `false`;
- the watchdog has bitten.

To facilitate using only a single fault source, `ok-in` and `fault-in` are both set to the non-fault-causing value when no signal is connected. For estop-latch to ever be able to signal a fault, at least one of these inputs must be connected.

**Usage in HAL**
Typically, the software EStop is connected to `ok-in`, `iocontrol.0.user-request-enable` is connected to `reset`, and `ok-out` is connected to `iocontrol.0.emc-enable-in`, so for example:
```
net estop-loopout iocontrol.0.emc-enable-in <= EMCO5_SPI.watchdog.ok_out
net estop-reset iocontrol.0.user-request-enable => EMCO5_SPI.watchdog.reset
net estop-user iocontrol.0.user-enable-out => EMCO5_SPI.watchdog.ok-in
```

With this basic setup, the enable function will go LOW when the user disables Linux-CNC. **NOTE: ** the heartbeat will always be beating as long the communication is OK.

Peter-van-Tol, 2024-11-11:
Working with an EStop is now supported:
``` json
{
    "board_name": "EMCO5_SPI",
    "board_type": "HUB75HAT v8.0",
    "clock_frequency": 40000000,
    "connection": {
        "connection_type": "spi",
            "mosi": "spi:0",
            "miso": "spi:1",
            "clk":  "spi:2",
            "cs_n": "spi:3"
    },
    "watchdog": {
        "estop": [
            {
                "name": "user-button",
                "pin": "user_btn:1",
                "trigger": false
            }
        ],
        "functions": [
            {
                "function_type": "enable",
                "pin": "ena:0"
            },
            {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true,
                "waveform": "heartbeat",
                "speed": 0.857
            }
        ]
    },
    "modules": [
    ]
}
```

The above configuration can be tested with the following script in `halrun`:
```
loadrt litexcnc connections="spidev:/dev/spidev0.0"
loadrt threads name1=test-thread period1=50000000
addf EMCO5_SPI.read test-thread
addf EMCO5_SPI.write test-thread
start
setp EMCO5_SPI.watchdog.reset 1
setp EMCO5_SPI.watchdog.reset 0
```

Above script will enable the watchdog and the board. One can simulate an E-Stop by pressing the user-button. The ENA LED will turn off and the board is stopped. The whole handling of E-Stop is now onboard. Even when communication gets lost at this very moment, the FPGA will go into E-Stop. 

When the board is in E-Stop mode, the HAL pin `fault-out` will become `true` and `ok-out` becomes `false`. Additionally, for each E-Stop defined in the config, extra pins are created in the config. These pins can be used to determine the source of the E-Stop.

All outputs are latched, i.e. the signals are only returned to their working state when the watchdog is reset:
 ```
setp EMCO5_SPI.watchdog.reset 1
setp EMCO5_SPI.watchdog.reset 0
```

**Work-in-progress:**
This is still work-in-progress. Only a single E-Stop is supported at this moment, but more flexibility will be added very soon.

Peter-van-Tol, 2024-11-12:
@chaotix- : The revamp is done. Could you test it?

chaotix-, 2024-11-13:
I the branch does not compile with the GPIO module and at least one pin configured as output.

```
{
    "board_name": "lxcnc",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 40000000,
    "connection": {
        "connection_type": "etherbone",
        "tx_delay": 0,
        "ip_address": "10.0.0.10",
        "mac_address": "0x10e2d5000000"
    },
    "watchdog": {
	"estop": [
            {
                "name": "user-button",
                "pin": "user_btn:1",
                "trigger": false
            }
        ],
	"functions": [
            {
                "function_type": "enable",
                "pin": "j1:0"
            }, {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true,
                "waveform": "heartbeat",
                "speed": 0.857
            }
        ]
    },
    "modules": [
        {
            "module_type": "gpio",
            "instances": [
		{"direction": "out", "pin":"j7:0",  "name":"output1" }
            ]
        }
    ]
}
```

This results in the following traceback:
```
INFO:ECP5PLL:Creating ECP5PLL.
INFO:ECP5PLL:Registering Single Ended ClkIn of 25.00MHz.
INFO:ECP5PLL:Creating ClkOut0 sys of 40.00MHz (+-10000.00ppm).
INFO:ECP5PLL:Creating ClkOut1 sys_ps of 40.00MHz (+-10000.00ppm).
INFO:SoCBusHandler:master0 added as Bus Master.
Traceback (most recent call last):
  File "/home/lars/litexcnc/bin/litexcnc", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
[... 70 lines cut ...]
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/cli/build_firmware.py", line 50, in cli
    soc = firmware_config.generate()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/soc.py", line 109, in generate
    watchdog = self.watchdog.create_from_config(soc)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/config/modules/watchdog.py", line 142, in create_from_config
    return WatchDogModule.create_from_config(soc, self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/watchdog.py", line 320, in create_from_config
    estop_config=WatchDogModule.EStopConfig(soc, config),
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/litexcnc/lib/python3.11/site-packages/litexcnc/firmware/watchdog.py", line 169, in __init__
    self.pads = soc.platform.request_all("estop")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/toolchain/litex/litex/litex/build/generic_platform.py", line 322, in request_all
    return self.constraint_manager.request_all(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lars/toolchain/litex/litex/litex/build/generic_platform.py", line 233, in request_all
    raise ValueError
ValueError
```

Peter-van-Tol, 2024-11-13:
Thanks for the review.

The issues have been resolved:
- Removing the E-Stop all together was a blind spot, good find 👍 .
- There is a connection between the watchdog and the modules which have an output (GPIO-out, PWM, Stepgen). The interface was changed, but these changes were not merged.

I've tested with GPIO and PWM myself. Could you also test it with these improvements.

chaotix-, 2024-11-14:
I tried it today for a little bit.

Things I observed:
- When I pull the network cable the heartbeat off the onboard-LED stops immediately, but LinuxCNC does not trip the watchdog. The watchdog is tripped a second or so after I plug in the network cable again. Is there some way to indicate to LinuxCNC that the connection is broken? At the very least I want to turn off the spindle which is controlled via Modbus and not via the Colorlight-board. I think this was not a problem before the watchdog rewrite, but I have not tested this against `main` today.
- When I pull the plug and insert it again I get the message regarding the watchdog. When I start LinuxCNC again and pull and insert the plug again the watchdog does not always trip.
- Sometimes the X, Y and Z axis started to move. I think when plugging in the cable and when the watchdog was not triggered in LinuxCNC. (I'm not completely sure about the timing and what causes what. I just saw the axis move several time sometimes with a joint following error. I did not have much time today to test this.)

I will try to do some more tests in the next couple of days.

Peter-van-Tol, 2024-11-14:
Thanks for addressing some more issues!

> When I pull the network cable the heartbeat off the onboard-LED stops immediately, but LinuxCNC does not trip the watchdog. The watchdog is tripped a second or so after I plug in the network cable [...]

This behavior didn't change; there is no counter on dropped packets. Until now I focused on making sure the FPGA is in a safe-state when the connection is lost. I.e. when the wathcdog is tripped, all outputs are disabled. Didn't work yet on the driver to inform LinuxCNC on the connection status. 

> When I pull the plug and insert it again I get the message regarding the watchdog. When I start the machine again and pull and insert the plug again the watchdog does not always trip.

This is strange at first glance. When LinuxCNC is started, the card is reset so it will be in a known state. The behavior should not be depending on the state of the card. What I think what happens is:
- the watchdog has bitten;
- LinuxCNC does not know the state of the FPGA due to communication problems;
- the cable is plugged in between reading the FPGA (which is missed) and writing data to the FPGA (which will re-enable the watchdog)
- LinuxCNC misses the biting watchdog.

Above illustrates that it is important that LinuxCNC errors when communication is lost. Or the communication protocol should latch the `watchdog.has_bitten` on the FPGA as well.

> Sometimes the X, Y and Z axis started to move. I think when plugging in the cable and when the watchdog was not triggered in LinuxCNC. (I'm not completely sure about the timing and what causes what. [...]

I think this boils down to the same problem of LinuxCNC not getting the EStop signal if the timing of pluging it back in is bad.

Peter-van-Tol, 2024-11-14:
Question: when pulling the plug on your ethernet connection, do you get the error `Unexpected read length ...`. 

If so, I found the hook for the LinuxCNC `fault-out`.

chaotix-, 2024-11-15:
> Question: when pulling the plug on your ethernet connection, do you get the error `Unexpected read length ...`.

Yes I do get `Unexpected read length: -1, expected 68` multiple times.

chaotix-, 2024-11-15:
With this config the watchdog does not trigger anymore when pressing the onboard button:

```
{
    "board_name": "lxcnc",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 40000000,
    "connection": {
        "connection_type": "etherbone",
        "tx_delay": 0,
        "ip_address": "10.0.0.10",
        "mac_address": "0x10e2d5000000"
    },
    "watchdog": {
	"estop": [
            {
                "name": "user-button",
                "pin": "user_btn:1",
                "trigger": false
            }
        ],
	"functions": [
            {
                "function_type": "enable",
                "pin": "j1:0"
            },
            {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true,
                "waveform": "heartbeat",
                "speed": 0.857
            }
        ]
    },
    "modules": [
    ]
}
```

Even the onboard LED is pulsating happily without a care in the world when pressing the button.

chaotix-, 2024-11-15:
With a more complete JSON-config file LinuxCNC starts, but pressing the onboard button does not trip the watchdog.
Even the onboard LED just continues pulsating.

```
{
    "board_name": "lxcnc",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 40000000,
    "connection": {
        "connection_type": "etherbone",
        "tx_delay": 0,
        "ip_address": "10.0.0.10",
        "mac_address": "0x10e2d5000000"
    },
    "watchdog": {
	"estop": [
            {
                "name": "user-button",
                "pin": "user_btn:1",
                "trigger": false
            }
        ],
	"functions": [
            {
                "function_type": "enable",
                "pin": "j1:0"
            },
            {
                "function_type": "heartbeat",
                "pin": "user_led:1",
                "invert_output": true,
                "waveform": "heartbeat",
                "speed": 0.857
            }
        ]
    },
    "modules": [
        {
            "module_type": "gpio",
            "instances": [
		{"direction": "out", "pin":"j8:14", "name":"stepgen.enable" },
		{"direction": "out", "pin":"j7:0",  "name":"output1" },
		{"direction": "out", "pin":"j7:4",  "name":"output2" },
		{"direction": "out", "pin":"j7:2",  "name":"output3" },
                {"direction": "in",  "pin":"j9:0",  "name":"alarm-x" },
                {"direction": "in",  "pin":"j9:1",  "name":"alarm-y" },
                {"direction": "in",  "pin":"j9:2",  "name":"alarm-z" },
                {"direction": "in",  "pin":"j9:4",  "name":"home-x" },
                {"direction": "in",  "pin":"j9:5",  "name":"home-y" },
                {"direction": "in",  "pin":"j9:6",  "name":"home-z" },
                {"direction": "in",  "pin":"j10:0", "name":"input1" },
                {"direction": "in",  "pin":"j10:1", "name":"input2" },
                {"direction": "in",  "pin":"j10:2", "name":"input3" },
                {"direction": "in",  "pin":"j10:4", "name":"input4" },
                {"direction": "in",  "pin":"j10:5", "name":"input5" },
                {"direction": "in",  "pin":"j10:6", "name":"input6" }
            ]
        }, {
            "module_type": "stepgen",
            "instances": [
                {
                    "pins" : {
                        "stepgen_type": "step_dir",
                        "step_pin": "j8:0",
                        "dir_pin": "j8:2"
                    },
                    "soft_stop": true
                }, {
                    "pins" : {
                        "stepgen_type": "step_dir",
                        "step_pin": "j8:1",
                        "dir_pin": "j8:5"
                    },
                    "soft_stop": true
                }, {
                    "pins" : {
                        "stepgen_type": "step_dir",
                        "step_pin": "j8:4",
                        "dir_pin": "j8:6"
                    },
                    "soft_stop": true
                }
            ]
        }
    ]
}
```

Peter-van-Tol, 2024-11-15:
Ok, I've implemented a counter on the error message `Unexpected read length: -1, expected 68`. This will not take this message away, but you can see how many message have been missed. The number of missed message will be read by the watch-dog, so it can trip.

> With a more complete JSON-config file LinuxCNC starts, but pressing the onboard button does not trip the watchdog.
Even the onboard LED just continues pulsating.

How the watchdog works when pressing an E-Stop:
- the blinking LED (function heartbeat) will be on as long the communication is on.
- the message `watchdog has bitten` only shows when a timeout has occurred. I'm considering to remove the message, because the other causes of the watchdog tripping don't show a message.
- when the watchdog trips (due to E-Stop or communication timeout), the `fault-out` will be HIGH and `ok-out` will be LOW.
- to determine the source of tripping the watchdog, one can monitor the pins `watchdog.has_bitten` and `watchdog.estop.##.fault-out`.

> With this config LinuxCNC does not start:

The reason this happens is:
- when you don't specify any modules, there is no config-data to read;
- it requests an empty read to the FPGA, which it does not answer;
- because there is no answer, the initialisation fails and LinuxCNC / halrun won't start.

I will put a safeguard in place to prevent 0-byte reads.

Again, a very valuable contribution!

Peter-van-Tol, 2024-11-17:
I've prepared a new version of the driver for the watchdog, which actively monitors reading data from the FPGA. The following cases are covered:
- when the reads fail, the LinuxCNC driver will set `fault-out` of the watchdog to TRUE. This signal can be used to halt LinuxCNC. The driver will try to send the fault-condition to the FPGA, just in case it listens
- when the writes fail, the FPGA will not receive updates and will have a time-out (watchdog bites). This will stop all modules with outputs on the card (GPIO out, PWM and stepgen). It will try to send the `time-out` signal to LinuxCNC in case it is still receiving messages. 

When you pull the plug, the FPGA should experience a timeout _and_ the watchdog should set `fault-out` to TRUE.

@chaotix- : My system is configured as SPI, so it is not easy to 'pull the plug'. Could you test whether the system gives the desired effects on the Ethernet version. Also, the issue for the case without modules defined has been resolved.

chaotix-, 2024-11-19:
I tested the watchdog without any modules and in my tests it worked really well.

This is the HAL-config I used which shows how to tie in more estop-sources (maybe the vfd). If you don't mind starting the machine twice you can remove the `timedelay` component. When writing to `estop-latch.0.reset` it will trip again immediately as `[LITEXCNC](NAME).watchdog.fault_out` is still `1` on the first reset.

```
loadrt [EMCMOT]EMCMOT servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[KINS]JOINTS
loadrt litexcnc connections=[LITEXCNC](CONNECTION)
source cncpi-5a-75e_v8.2-watchdog/alias.hal
loadrt lut5 names=lut5-estop
loadrt timedelay names=timedelay-estop
loadrt estop_latch

# Add the functions to the thread
addf [LITEXCNC](NAME).read servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf lut5-estop servo-thread
addf estop-latch.0 servo-thread
addf [LITEXCNC](NAME).write servo-thread
addf timedelay-estop servo-thread

setp timedelay-estop.on-delay 0.0001
setp timedelay-estop.off-delay 0.0001

# Setup the watchdog (assuming servo-thread period of 1000000 ns (1 kHz))
setp [LITEXCNC](NAME).watchdog.timeout_ns 1500000

# Tie the watchdog into the E-STOP chain
# i0 = 0xaaaaaaaa;i1 = 0xcccccccc;i2 = 0xf0f0f0f0;i3 = 0xff00ff00;i4 = 0xffff0000;print('0x%08x' % (i0 | i1 | i2 | i3 | i4))
setp lut5-estop.function 0xfffffffe	# Bit0 OR Bit1 OR Bit2 OR Bit3 OR Bit4
net estop-watchdog lut5-estop.in-0 <= [LITEXCNC](NAME).watchdog.fault_out
#net estop-user_btn lut5-estop.in-1 <= [LITEXCNC](NAME).gpio.user_btn.in-not

net estop-loopout iocontrol.0.emc-enable-in <= estop-latch.0.ok-out
net estop-loopin iocontrol.0.user-enable-out => estop-latch.0.ok-in
net estop-reset <= iocontrol.0.user-request-enable
net estop-reset => [LITEXCNC](NAME).watchdog.reset
net estop-reset => timedelay-estop.in
# Reset estop-latch after LITEXCNC watchdog had a chance to reset.
net estop-reset-delayed timedelay-estop.out => estop-latch.0.reset
net estop-lut5 estop-latch.0.fault-in <= lut5-estop.out

# More sources for E-stop (such as a GPIO-in) can be added to this E-Stop circuit by attaching them to estop-latch.0.fault-in-X
```

Peter-van-Tol, 2024-11-19:
> When I remove gpio and stepgen from modules (so without any defined modules at all) LinuxCNC does start.

I will check why it does not accept modules...

> This is the HAL-config I used which shows how to tie in more estop-sources (maybe the vfd). If you don't mind starting the machine twice you can remove the timedelay component.

Maybe we can do without the delay, when we use the latches is serie. The order is important in this case. Based on your example:
``` 
loadrt [EMCMOT]EMCMOT servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[KINS]JOINTS
loadrt litexcnc connections=[LITEXCNC](CONNECTION)
source cncpi-5a-75e_v8.2-watchdog/alias.hal
loadrt estop_latch

# Add the functions to the thread
addf [LITEXCNC](NAME).read servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf estop-latch.0 servo-thread
addf [LITEXCNC](NAME).write servo-thread

# Create singal for resetting the latches
net estop-reset iocontrol.0.user-request-enable

# Tie in external e-stops (for example FPGA)
net estop-user iocontrol.0.user-enable-out => estop_latch.0.ok-in
net estop-reset => estop_latch.0.ok-in
# net latch-0-fault-in estop_latch.0.fault-in <= <PIN OF EXTERNAL ESTOP>
net latch-0-ok-out estop_latch.0.ok-out

# Tie in the watchdog
net latch-0-ok-out => [LITEXCNC](NAME).watchdog.ok-in
net estop-loopout iocontrol.0.emc-enable-in <= [LITEXCNC](NAME).watchdog.ok_out
net estop-reset => [LITEXCNC](NAME).watchdog.reset
```

When resetting:
- first the `estop_latch` is reset, which is only looking at the external E-Stop (for example your VFD).
- When this latch can be freed, the `ok-in` signal is set to TRUE and this value is written to the FPGA. 
- If everything is fine on the FPGA, this latch will be freed as well.
- At the start of the next period, the state of the FPGA is read and the `iocontrol` is enabled.

One can add more latches when required to make a ladder. The last latch will always be the watchdog on the FPGA.

chaotix-, 2024-11-19:
Driver now loads with modules enabled.

Your estop chain also works (after correcting some `_` and `-` confusion). Maybe you could use `-` in the pin names in LiteX-CNC to use the same convention as other modules in LinuxCNC.

When using several estop-inputs I think I would still prefer the `lut5`-solution instead of a ladder but this is personal preference and at the moment I don't know what second input I would need on the LinuxCNC-side.

I also tested two simultaneous estop-inputs on the FPGA side (the onboard button and a HUB75-input pin) with `"trigger": true` as well as `"trigger": false`. Worked like a charm. Pulling the ethernet plug also tripped the estop on the LinuxCNC-side without any noticeable delay.

Peter-van-Tol, 2024-11-19:
Within LinuxCNC there are many roads which lead to Rome. You can use `lut5` to minimise the ladder, you can use a ladder of latches, my lathe works with a custom component due to some interlocking and you can also go to classic ladder. In all cases, if the FPGA latch or watchdog is the last part of the chain, things work without the need of a delay.

Your remark about `_` and `-`: I've checked all pins to obey the LinuxCNC covention, updated the alias generator and documentation. There are no more `_` in the pin names exported by LitexCNC.

If you could test this last change, then I will close this issue and merge it to `main`. Thank you so much for your patience and help with testing!

Peter-van-Tol, 2024-11-19:
That was the last `_`. I only had changed the first hyphen.

Time to merge I think?

chaotix-, 2024-11-20:
Looks good to me. (Famous last words. :-))

At least my last test did not show any more problems.

Peter-van-Tol, 2024-11-20:
Famous last words ... We can always open an new issue. 

I will finalize the documentation and close this issue over the weekend. Thanks a lot!

closed 2025-01-09 by Peter-van-Tol