Peter-van-Tol/LiteX-CNC #120: [Bug]: Encoder readings wrong

marland4a, 2024-11-08:
### Description

I have a machine with 4 axis (dual Y), running with PWM-controlled servos and encoders.
I built the attached configuration file with no errors and used openFPGALoader to flash the bit file because openOCD somehow did not work.
I can start LinuxCNC and I can use the e-stop, the limit switches and the PWM outputs.
But the encoder readings are not working at all.

I checked pretty much any example configuration and bug report I could find.
In the hal meter, the raw-counts value of each encoder changes when moving the axis, but it only switches between 0 and 1 or 0 and -1 on some axis. The position, counts etc stay at 0.

I originally used the 1.3.0-dev3(?) built from the main branch because I need up/down PWM.
As it didn't work, I downgraded to 1.2.4 installed via pip and reinstalled the driver and the toolchain, but it shows the same behavior. Thats the reason for the even more PWM instances, since I moved the up/down part into the LinuxCNC HAL file.
I confirmed with an oszilloscope that the encoders work and that I used the correct pins. I exchanged the buffers with 74vc245 on the input pins and confirmed they work by testing every pin with a multimeter.

One fix I tried yielded very interesting results:
I added a single stepgen output to j8:5 and j8:6 in case there are some dependencies.
After that, the first encoder seems to contain the wallclock.ticks_lsb value, while the actual wallclock.ticks_lsb value stays 0.
The counts, position, velocity etc values of the encoder are increasing as well, so I guess the LinuxCNC driver works.

So my guess is that there is some problem with the communication, maybe because of the many PWM signals?

Current config with only XYYZ pwm and encoders:
[firmware_v1_config_onlyXYYZ.json](https://github.com/user-attachments/files/17683651/firmware_v1_config_onlyXYYZ.json)
Resulting gateware folder:
[gateware.zip](https://github.com/user-attachments/files/17683720/gateware.zip)

Complete LinuxCNC config:
[alias.hal.txt](https://github.com/user-attachments/files/17683722/alias.hal.txt)
[alias2.hal.txt](https://github.com/user-attachments/files/17683723/alias2.hal.txt)
[custom.hal.txt](https://github.com/user-attachments/files/17683725/custom.hal.txt)
[LitexCNC_Colorlight75E_legacy.ini.txt](https://github.com/user-attachments/files/17683726/LitexCNC_Colorlight75E_legacy.ini.txt)
[LitexCNC_Colorlight75E_legacy.hal.txt](https://github.com/user-attachments/files/17683727/LitexCNC_Colorlight75E_legacy.hal.txt)

Thank you for your help

### Which FPGA are you using

- [ ] 5A-75B
- [X] 5A-75E
- [ ] 5A-75B / HUB75HAT
- [ ] other (please specify in issue)

### Python, Litex-CNC and LinuxCNC & OS Version

```Text
LinuxCNC 2.9.3 on Debian 12 preempt-rt, installed from the official image (https://www.linuxcnc.org/iso/linuxcnc_2.9.3-amd64.hybrid.iso)
Python 3.11.2
LitexCNC 1.2.4 installed via pip
The same issues happen when running 1.3.0 built from the main branch
```

Peter-van-Tol, 2024-11-08:
As you require the new style PWM, could you please install LitexCNC from `main`? 
```shell
pip install git+https://github.com/Peter-van-Tol/LiteX-CNC
``` 

This will require to modify your configuration to include the new style for `pwm`, see an example below (partial example):
```json
     {
         "module_type": "pwm",
         "instances": [
             {
                 "pins": {
                     "output_type": "updown",
                     "up": "j2:1",
                     "down": "j2:2"
                 }
             }, {
                 "pins": {
                     "output_type": "pwmdirection",
                     "pwm": "j2:4",
                     "direction": "j2:5"
                 }
             }, {
                 "pins": {
                     "output_type": "single",
                     "pwm": "j2:6"
                 }
             }, {
                 "pins": {
                     "output_type": "single",
                     "pwm": "j2:7"
                 }
             }
         ]
     }
```

Can you then try a minimal example to test your encoders, i.e.:
```json
{
    "board_name": "75E",
    "board_type": "5A-75E v6.0",
    "clock_frequency": 40000000,
    "connection": {
        "connection_type": "etherbone",
        "tx_delay": 0,
        "ip_address": "10.0.0.10",
        "mac_address": "0x10e2d5000001"
    },
    "watchdog": {
        "pin":"j1:9"
    },
    "modules": [
        {
            "module_type": "encoder",
            "instances" : [
                {"name": "enc_mot_X", "pin_A": "j10:0", "pin_B": "j10:1"},
                {"name": "enc_mot_Y1", "pin_A": "j11:0", "pin_B": "j11:1"},
                {"name": "enc_mot_Y2", "pin_A": "j11:4", "pin_B": "j11:5"},
                {"name": "enc_mot_Z", "pin_A": "j10:4", "pin_B": "j10:5"}
            ]
	}
    ]
}
```

When this example works, then you can add PWM and GPIO to pinpoint the error. Please let me know the results.

marland4a, 2024-11-09:
I was able to fix the issues today.
It was actually not a software issue, but a combination of two hardware problems. First, I used the wrong HUB75 pinout and some of the connectors I used for the encoders were faulty.

Anyway, thank you for your fast response and for pointing me into the right direction.

Peter-van-Tol, 2024-11-10:
Great it works now! If you need any help to get your system up and running, feel free to ask.

For future reference: counting is done within the FPGA, not in the software / LinuxCNC. A value which only shows between 0 and 1 or between 0 and -1 indicates a wiring issues.

closed 2024-11-10 by Peter-van-Tol