Peter-van-Tol/LiteX-CNC #132: Add firmware update via ethernet

romanetz, 2025-02-07:
Hi everybody I'd like to suggest firmware update through ether bone interface. This can be done in two ways: first way is to map flash onto CSRs and use etherbone protocol
And second way is to add vexriscv core and onboard dram into the SoC. Thus, implementing a buffered firmware update. I have some code regarding implementation of both approaches and will share it later, as soon as I'll have access to my development machine

Peter-van-Tol, 2025-02-07:
Please share your code. I will definately add this to the code.

Fits nicely eith the refactoring required for supporting more boards.

morganrallen, 2025-07-01:
The idea of a soft core has popped into my head a couple of times for scripting/automation. Seems VexRiscv can even run Linux (with Python!), that would open up tons of interesting opportunities.