Peter-van-Tol/LiteX-CNC #140: No Port open on 1234

Gamer08YT, 2025-09-13:
When I flash the latest firmware on my 5A-75B v8.0, the NIC gets display in my DHCP Server, but a port scan on 1234 tcp/udp shows no open port.

romanetz, 2025-09-13:
Liteeth neither provides either DHCP support nor it responds to any random
UDP/TCP packet. ICMP ping and ARP is supported.

Gamer08YT, 2025-09-14:
Sadly it throws an error, but the device is pingable.

```(venv) root@ByteCNC:/home/jaxnprivate/Desktop/ByteCNC# ping 192.168.1.224
PING 192.168.1.224 (192.168.1.224) 56(84) bytes of data.
64 bytes from 192.168.1.224: icmp_seq=1 ttl=128 time=0.843 ms
64 bytes from 192.168.1.224: icmp_seq=2 ttl=128 time=0.720 ms
64 bytes from 192.168.1.224: icmp_seq=3 ttl=128 time=0.827 ms
^C
--- 192.168.1.224 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2041ms
rtt min/avg/max/mdev = 0.720/0.796/0.843/0.054 ms
(venv) root@ByteCNC:/home/jaxnprivate/Desktop/ByteCNC# litex_server --udp --udp-ip=192.168.1.224
[CommUDP] ip: 192.168.1.224 / port: 1234 / tcp port: 1234
Traceback (most recent call last):
  File "/home/jaxnprivate/Desktop/ByteCNC/venv/bin/litex_server", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'litex_server')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/litex/litex/litex/tools/litex_server.py", line 285, in main
    server.open()
  File "/opt/litex/litex/litex/tools/litex_server.py", line 91, in open
    self.comm.open()
  File "/opt/litex/litex/litex/tools/remote/comm_udp.py", line 35, in open
    self.probe(self.server, self.port)
  File "/opt/litex/litex/litex/tools/remote/comm_udp.py", line 67, in probe
    raise Exception(f"Unable to probe Etherbone server at {self.server}.")
Exception: Unable to probe Etherbone server at 192.168.1.224.
(venv) root@ByteCNC:/home/jaxnprivate/Desktop/ByteCNC# 
```

[image]

Gamer08YT, 2025-09-14:
Totally my problem, sorry for wasting your time ;)

Since I've been waiting for my Thincenter for two weeks, I set up a VM. Apparently there were problems with the “NAT” adapter, so I set the NIC as a bridge directly to my PC's NIC and lo and behold, HAL works :P

Thanks for your help, your commands were very helpful and are somehow missing from the docs. It would be cool to have a little more help like that.

Let's see if I can contribute something once I'm a little more familiar with the card.

Once again, we see that the pure PING command is not reliable. You'd think Ethernet could be so simple ;)

closed 2025-09-14 by Gamer08YT